ARMmbed/mbed-os #1262: Windows 8 Compatability

jamesadevine, 2015-07-23:
The test suite `singletest.py` does not appear to work with windows 8.

The issue seems to be with Serial, which all of the tests rely on. 

After the test suite resets the target, the response received on serial is as follows:

`HOST: Unknown property:`

When viewed through a serial terminal:

```
{{timeout;5}}
{{host_test_name;hello_auto}}
{{description;Hello World}}
{{test_id;MBED_10}}
{{start}}
Hello World
[00]{{timeout;5}}
{{host_test_name;hello_auto}}
{{description;Hello World}}
{{test_id;MBED_10}}
{{start}}
Hello World
```

Replicating the same build environment on a windows 7 machine, all tests run successfully.

It could be a symptom of the virtual machine however...

I am running windows 8 on the VMware box, but my serial terminal can see it, why can't the script?

sg-, 2015-07-23:
is the serial driver installed on the windows 8 machine?

jamesadevine, 2015-07-23:
@sg- the one referred to here? https://developer.mbed.org/handbook/Windows-serial-configuration

bridadan, 2015-07-23:
Also to clarify, you don't have the terminal and the test suite running at the same time right? I believe the test suite will try to "own" the serial port and if the terminal is running it won't be able to run correctly.

jamesadevine, 2015-07-23:
@bridadan I do not, it would give a different error if that were the case.

sg-, 2015-07-23:
@jamesadevine yes

jamesadevine, 2015-07-23:
@sg- it is installed

0xc0170, 2015-07-29:
@PrzemekWirkus do we have any machine with this OS which causes this issue?

PrzemekWirkus, 2015-07-30:
@0xc0170 No we do not, and I do not think we "support" Windows 8.
For now we are working with Windows 7, Ubuntu LTSs and for some tools MacOS (Darwin).

PrzemekWirkus, 2015-07-30:
@jamesadevine We do not support test suit running on VMs yet!

closed 2015-08-13 by PrzemekWirkus