ARMmbed/mbed-os #1279: [nRF51] System off & Pin 31 disabled

xcrespo, 2015-08-04:
I found two commented lines when using mbed for the nRF51822 without any apparent reason which cause some troubles.
* [The first one](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51822_MKIT/PinNames.h#L64) should enable system off mode [(Nordic ref. on this topic)](https://github.com/NordicSemiconductor/nrf51-powerdown-examples).
* [The last one](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/sleep.c#L31) disables the usage of pin number 31.

I'd like to know why are they commented out before I submit a pull request.

jrobeson, 2015-09-25:
ping mbed maintainers?

jrobeson, 2015-10-05:
maybe submitting a pull request would get a better response as to why it is wrong to do so.

crespum, 2016-02-09:
I've already obtained a response for the absence of the pin 31 in #1476. I suggest to create a new issue regarding just to the system off mode which has not been clarified.

closed 2016-02-09 by crespum