ARMmbed/mbed-os #1284: USB device - Current consumption stays high before 1st connect

amveeq, 2015-08-07:
We've been working on the USB Device for a while and we noticed that there is some shortcomings in the USB device library which result in the high current consumption before (any class) USB device is connected for the 1st time. There has been already reported couple of issues “USB Serial constructor blocking" (https://github.com/mbedmicro/mbed/issues/53) and “Allow USBDevice::connect() to be non-blocking" (https://github.com/mbedmicro/mbed/issues/169) with some level of fixes.
Related to the issues mentioned there is also another potential way to correct the issues. Constructing the Serial (or MSD class) object a bit differently, current consumption can be limited in the main function: USBSerial (VID, PID, REL, **false**). This is possible with Serial and MSD classes as they support non-blocking construction.
**However this still leaves Audio and HID classes with the current consumption issue as they do not support non-blocking construction**.

c1728p9, 2015-08-21:
Hi @amveeq , you should be able to create a HID class without extra current draw since the last parameter in the constructor is whether to call connect in the constructor.  If you set this to false and then call connect(false) later, it should fix your current problem.  Let me know if I missed something in your post.  Also, it looks like the blocking inside the audio class is a problem.

amveeq, 2015-08-24:
@c1728p9, yes HID class seems also to have the non-blocking capability but audio don't so that still remains to be a problem.

closed 2017-10-27 by ChiefBureaucraticOfficer