ARMmbed/mbed-os #1295: LED error patterns

Sissors, 2015-08-13:
Since Martin wanted me to post it here, and otherwise we never get the 200 open issues:

The two original mbeds have a very nice, very obvious, error pattern. But that only works for the 4 LEDs on that board, while pretty much no other board has 4 seperate LEDs. Some of the other mbeds don't even have error LEDs (although they seem to be being added now). Some of these are red, which you could say indicates an error. But enough simply have no red LEDs, and only have a green LED for example. On all boards with RGB LEDs (Freescale to start with), it is pretty much random which color it blinks, depending on the state of the other colors of the RGB LED. It can blink red, but it might very well blink purple-blue (if the blue LED is enabled), or yellow-white (if green/blue LEDs are enabled).

Now add to it that blinking an LED isn't exactly special: The original mbed pattern was special, but if a green LED is blinking I don't immediatly think: "Ah an error". I assume it is just blinking because half my programs have random LEDs blinking. 

So the main problem is a lack of consistency. Try to describe how a new user knows an error happened in one sentence, good luck. "A certain LED blinks, or multiple LED blinks, something blinks, that tells you an error happened".

Solution I see: Make a distinct blinking pattern (for example 4 times fast, 4 times slow. As long as it is something that you notice clearly as out of the ordinary). Document this somewhere where new users will actually see it (!!!), new users especially will run often into this, everytime they use wrong pins for a peripheral. And use this same error patterns for every board. Just blink all LEDs in this pattern, then you don't need different defines for each board (and then also remove the target specific disable_irq, just disable them for all). 

And if you really want you can let the two original mbeds keep their funky blinking pattern, but also in this new speed pattern, so as example 4 times fast, 4 times slow).

sg-, 2015-08-13:
+1 to consistency and distinct pattern. The existing pattern is about the same speed as blinky :( 

I think this should be a design challenge. Blinky olympics 2015

NewLunarFire, 2015-08-13:
Great idea. I agree that we should keep the original patterns for the first two mbed boards (the LPC1768 and what is the second one?). As soon as we are decided on a pattern (4 fast 4 slow seems fine to me), we should start implementing them on all of the boards. It should be simple enough to do, as you said blinking LEDs is nothing really special and it's almost always the first assignment for electronics students (I must have done dozens of LED chasers during my college years).

star297, 2015-08-13:
That's a blinking good idea Erik ! Four rapid flashes followed by a pause, a bit like emergency services vehicles could be a good sequence.
Mind you it probably wouldn't apply to me, I never make mistakes in my code :)

I did notice there is no error indication on the EFM32's and we need it on the Teensy3.1 as well.

PR #4920 (2017-08-24): Modify LED error sequence to be more recognisable
Reopening this PR from the [previous PR](https://github.com/ARMmbed/mbed-os/pull/4883).  I accidentally closed the previous one.

Unified the LED error sequence for all boards. Previously the LED sequence for boards with 4 LEDs was a patterned sequence, and for other boards, a fixed blinking LED. The new sequence is a repeating sequence of 4 fast pulses, followed by 4 slow pulses.

Fixes #1295 LED error patterns
closed 2017-08-24 by theotherjimmy