ARMmbed/mbed-os #1308: Sync command in test suite causes --parallel option to fail

bridadan, 2015-08-24:
```
Executing 'python host_test.py -d /media/bridan01/MBED5 -f "/home/bridan01/dev/myfork/mbed/build/test/LPC1768/GCC_ARM/MBED_26/div.bin" -p /dev/ttyACM1 -t 10 -C 1.5 -m LPC1768'
Test::Output::Start
.MBED: Instrumentation: "/dev/ttyACM1" and disk: "/media/bridan01/MBED5"
HOST: Copy image onto target...
cp: failed to access ‘/media/bridan01/MBED4/timeout.bin’: Permission denied
.Test::Output::Finish
TargetTest::K64F[0]::GCC_ARM::MBED_24::Timeout Int us [TIMEOUT] in 30.31 of 15 sec
Building project TIME_US (K64F, GCC_ARM)
Executing 'python host_test.py -d /media/bridan01/MBED4 -f "/home/bridan01/dev/myfork/mbed/build/test/K64F/GCC_ARM/MBED_25/time_us.bin" -p /dev/ttyACM0 -t 15 -C 4 -m K64F'
Test::Output::Start
cp: failed to access ‘/media/bridan01/MBED4/time_us.bin’: Permission denied
MBED: Instrumentation: "/dev/ttyACM0" and disk: "/media/bridan01/MBED4"
HOST: Copy image onto target...
Test::Output::Finish
```

Note the lines where cp fails with the error `cp: failed to access ... Permission denied`.

I've also gotten mbeds into a sort of deadlock, where every test after a certain point times out.

This also happens to a lesser degree when running multiple instances of singletest.py in different terminals without using `--parallel`.

That being said, I haven't had an issue with it when testing multiple mbeds without the `--parallel` option. But it takes quite a while to test when you have lots of boards connected.

0xc0170, 2015-09-01:
cc @PrzemekWirkus

bridadan, 2015-09-01:
After more investigation, I've found the reason why this is actually failing. When testing in parallel, the mbeds don't always mount to the same directory. Mbed-ls is only used at the beginning of the testing process, so it was trying to flash incorrect binaries. I've done some work so that mbed-ls is consulted before each test run, I'll try to get that into a PR soon.

closed 2015-09-01 by bridadan