ARMmbed/mbed-os #1328: MBED and mbedOS relationship?

anpilog, 2015-09-10:
Hi,

With releasing mbedOS we got code duplication for target platforms...
Fix me if I'm wrong but now mbed got split into:
- target
- hal
- drivers

Where is going to be mainstream work?
In https://github.com/mbedmicro or https://github.com/ARMmbed repositories?

Regards

kylemanna, 2015-12-01:
I'm curious as to whether or not the CMSIS-RTOS API will be implemented for mbedOS.

0xc0170, 2016-01-21:
Hello,

sorry for getting back with "small delay".

> Where is going to be mainstream work?
In https://github.com/mbedmicro or https://github.com/ARMmbed repositories?

As you could have noticed, most of the new development is happening for mbed OS.

> I'm curious as to whether or not the CMSIS-RTOS API will be implemented for mbedOS.

Something you should ask on the mbed.com forum.

sg-, 2017-01-11:
mbed OS development is continuing in this repository.

closed 2017-01-11 by sg-