ARMmbed/mbed-os #1329: Nucleo F030 broken on mbed builds 102 and onwards.

campag, 2015-09-10:
Last revision that works is 101. After that, even simple LED blinky code does not run. Paul Staron pinpointed the problem here:
https://developer.mbed.org/questions/60794/Nucleo-F030R8-simple-LED-blinky-not-work/#answer8553

0xc0170, 2015-09-10:
Paul points I believe to gcc arm startup, but you are having problem with the online ide or offline using keil, is that correct?

campag, 2015-09-10:
Correct, I'm using the online complier.

star297, 2015-09-10:
Sorry guys wrong file. Check the question link again for the fix. If I add that part back to the latest revision it works. I have a resolve for the RTC reset  bug on this target as well as the F401.

adustm, 2015-09-25:
Hello, 
I just used the online compiler, nucleo_blink_led example, and NUCLEO_F030R8 platform. 
I have tested mbed library revision 102 and 108. Both are blinking the led.
Same behavior with KEIL debugger
Kind regards

star297, 2015-09-25:
Is that with external crystal set up? I have checked now using MBED-SRC revision 627 (current) with external crystals fitted and it is not working. Was working (and still does) before the changes mentioned.

campag, 2015-09-25:
My Nucleo-F030R8 uses the HSI oscillator: no HSE crystal, nor MCO 8MHz solder bridges. I've just re-tested LED blinky code:

MBED-SRC 627 is not working for me. As Paul/star297 has identified, 582 is the last revision working here.

MBED lib 102-107 is not working here (can't see 108?), 101 is OK.

campag, 2015-09-28:
Have a blinking LED again after today's mbed-src 630 update. However, it's about 4x faster than it should be.

star297, 2015-09-28:
Just tried with rev. 634 and as you say running too fast.

0xc0170, 2016-01-21:
@dinau Is this fixed?

dinau, 2016-01-28:
Hi 0xc0170,
I've checked a simple LED blinky and printf() code using latest SDK srcs.
It's ok, works well.

dinau

closed 2016-01-28 by 0xc0170