ARMmbed/mbed-os #1335: Race condtion in LPC11CXX ADC

porkybrain, 2015-09-12:
https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC11XX_11CXX/analogin_api.c#L51

race condition on SYSAHBCLKCTRL  if different peripherals init on different threads.

Sissors, 2015-09-15:
While I guess on some MCUs (these not included by the way) bit banding could be used to change some operations to atomic (although it probably wouldn't help with clarity), I do think it is better to focus on possible race conditions outside the init phase. Although there I think the extra overhead if interrupts have to be toggled all the time is a significant (possible) issue.

However if we include initalization of different/several peripherals in different threads/interrupts the list of possible things which could go wrong gets so long, I think the best advice is to simply make sure in the user code it cannot switch to something else until you finish peripheral init.

porkybrain, 2015-09-15:
Ok fair enough, I guess one could just lock at the beginning of every init and unlock at the end. I guess if you issue guidance to the user that they should init everything before spinning up threads this would not be a problem. 

Otherwise I don't see an easy and practical way for a user to ensure synchronization without a lot of effort. Plus its not intuitive to have to lock when constructing an object, especially if its the only object of that kind in the whole project, so some users will no doubt overlook this. 

If you expect the user to init everything at start up then why not make most of the polymorphie static rather than using virtuals? Seems like a huge efficiency gain is possible there.

closed 2017-10-27 by ChiefBureaucraticOfficer