ARMmbed/mbed-os #1350: add nrf51 p26 & p27 to analog api

struempelix, 2015-09-24:
Issue: 

As of today nrf51 p26 & p27 can not be used as analog input as they are missing the analog api pin map definition. This might not be a problem for the DK as those pins are connected to the external XTAL anyways but on some prototyping boards like the xuntong ptr5188 these pins are the only way to use ADC.

So these two additional lines enable analog in on p26 & 27:

Lines to be added:
```
    {p26, ADC0_0, 1},
    {p27, ADC0_0, 2},
```
Current implementation:

```
static const PinMap PinMap_ADC[] = {
    {p1, ADC0_0, 4},
    {p2, ADC0_0, 8},
    {p3, ADC0_0, 16},
    {p4, ADC0_0, 32},
    {p5, ADC0_0, 64},
    {p6, ADC0_0, 128},
    {NC, NC, 0}
};
```

0xc0170, 2015-09-24:
Hi,

thanks for reporting. Are you planning to send PR with corrections to pins to enable those 2?

struempelix, 2015-09-24:
Hi, 
I've never done it before, but will give it a try. 
Cheers,
Jens

struempelix, 2015-09-24:
Ok, done! Hope I did it right.

jrobeson, 2015-10-09:
This issue can be closed now as the PR was merged

closed 2015-10-09 by 0xc0170