ARMmbed/mbed-os #1371: support for RedBearLabs BLENano hardware revision 1.5

jrobeson, 2015-10-04:
1.5 contains 32KB ram unlike the 16KB supported in the original. Should it be a new platform (like Teensy 3.0) or someting else. Pardon me if this isn't the right place to submit this.

jrobeson, 2015-10-09:
I think I'll be getting one of these in the next week, so I'd like to know how to name it.

0xc0170, 2015-10-09:
I believe a new platform is neeed . If up to 1.5 there's 16KB.

jrobeson, 2015-10-09:
I'll give it a try once I get one

jrobeson, 2015-12-24:
I have one of these, but I wonder if it's worth adding to mbed at this point. I'm already developing with yotta now.

alteredworlds, 2016-01-27:
Can anyone recommend a combination of library versions to get an offline build for mbed-classic working for the BLE Nano v1.5? I've been using embedXcode+ for Arduino (for a different board), but can't get a build working for mbed-classic. Out of interest I tried yotta/Docker (although sounds like this is not recommended for any potential production version for *many* months?) & also couldn't get any build working for the BLE Nano v1.5. Could easily be user error, of course...! :)

jrobeson, 2016-01-27:
I have blenano 1.5 working with yotta and mbed-os, but it should probably work for  mbed classic too.  However I'm using my own custom target and hal until the ARM folks publish theirs. It also requires one to edit https://github.com/ARMmbed/mbed-hal-nrf51822-mcu/ since the allowed targets are whitelisted there.

jrobeson, 2016-01-27:
NOTE: I haven't published my targets, because @AlessandroA said they will be publishing official ones soon.

alteredworlds, 2016-01-27:
OK, thanks for the info! 

Arduino 1.6.7 now seems to be working well for me (with board selection ‘BLE Nano(V1.5 32KB)’), after flashing the latest RedBear mk20 firmware from https://github.com/RedBearLab/nRF51822-Arduino <https://github.com/RedBearLab/nRF51822-Arduino>

(I had previously applied the firmware as documented at: https://developer.mbed.org/platforms/RedBearLab-BLE-Nano/ <https://developer.mbed.org/platforms/RedBearLab-BLE-Nano/>)

…so learning about this board can continue!

hotchpotch, 2016-09-27:
hi.

My BLE Nano ano V1.5 worked on mbed OS5.

- https://gist.github.com/hotchpotch/fe1d96ebabe3a46a41d5b5e9412811bd (Changed 16K to 32K)

closed 2017-10-27 by ChiefBureaucraticOfficer