ARMmbed/mbed-os #1378: multiple tickers stop working

andrewjfox, 2015-10-07:
Hi, I've read about previous issues with tickers being resolved, however I'm having trouble and want to confirm if ticker functions are all ok. I'm using nRF51422 based board and nRF51-DK platform on mbed to generate FOTA files.

Following is class used to drive RGB leds with single ticker for each led (there are 2 leds

```
#include "mbed.h"

class RGBColour
{
public:
    RGBColour(int r=0, int g=0, int b=0) {
        red = r;
        green = g;
        blue = b;
        brightness = 100;
    }    

    bool isOn(void) {
        if(!brightness) return false;
        if(!red && !green && !blue) return false;
        return true;
    }    
    
    int red;        // 0 - 100 %
    int green;
    int blue;
    int brightness; // 0 - 100 %
};

class RGBControl {
public:
#ifdef TARGET_DK
    const static int LED_ON    = 0;
    const static int LED_OFF   = 1;
#else
    const static int LED_ON    = 1;
    const static int LED_OFF   = 0;
#endif
    
    // 20ms PWM period with ? count resolution on RGB values
    // any faster and interrupts can't service in time
    const static float PWM_PERIOD = 0.025;
    const static int COUNTS_PER_PERIOD = 10;
    const static float SEC_PER_COUNT = (PWM_PERIOD / (float)COUNTS_PER_PERIOD);
    
    RGBControl(PinName red, PinName green, PinName blue) : _red(red), _green(green), _blue(blue) {  
        rgb(RGBColour(0,0,0)); 
    }
     
    /** @param RGB values
    */        
    void rgb(RGBColour c) {        
        _ticker.detach();
        _red = LED_OFF;
        _green = LED_OFF;
        _blue = LED_OFF;
        
        if(c.isOn()) {
            _tickerCompare[0] = (((c.red * c.brightness) / 100) * COUNTS_PER_PERIOD) / 100;
            _tickerCompare[1] = (((c.green * c.brightness) / 100) * COUNTS_PER_PERIOD) / 100;
            _tickerCompare[2] = (((c.blue * c.brightness) / 100) * COUNTS_PER_PERIOD) / 100;

            _tickerCount = COUNTS_PER_PERIOD;
            _ticker.attach(this, &RGBControl::tickerCallback, SEC_PER_COUNT);
        }    
    }
   
private:
    DigitalOut _red;
    DigitalOut _green;
    DigitalOut _blue;
    Ticker _ticker;
    int _tickerCount;
    int _tickerCompare[3];
   
    void tickerCallback(void)
    {
        if(++_tickerCount >= COUNTS_PER_PERIOD) { 
            _tickerCount=0;
            
            if(_tickerCompare[0]) _red = LED_ON;
            if(_tickerCompare[1]) _green = LED_ON;
            if(_tickerCompare[2]) _blue = LED_ON;
        }    
        
        if(_tickerCount == _tickerCompare[0]) _red = LED_OFF;
        if(_tickerCount == _tickerCompare[1]) _green = LED_OFF;
        if(_tickerCount == _tickerCompare[2]) _blue = LED_OFF;
    }  
};

```

if I drive a single LED, all good, if I drive 2 leds simultaneously, both of the tickers stop after several seconds.
```
RGBControl led1(LED1_RED,LED1_GREEN,LED1_BLUE);
RGBControl led2(LED2_RED,LED2_GREEN,LED2_BLUE);
...

// call this every second, increments through colour table
led1.rgb(colour[currentColour]);
if(++currentColour >= COLOUR_MAX) currentColour = COLOUR_BLACK;
led2.rgb(colour[currentColour]);

```

Is there an issue with my code, or is it possibly an unresolved ticker issue?

Andrew

anpilog, 2015-10-07:
Hi,

I guess you are running Ticker at too high frequency.
NRF51 platform use RTC for us_ticker and it's just 32kHz.
Besides there was old issue with us_tick_timer for NRF51 and some STM32 boards when us_ticker stuck under high load.
As I remember the issue on NRF51 target was resolved by using SDK API for RTC.

Anyway I would suggest to do high load test for us_ticker first.

Regards

andrewjfox, 2015-10-07:
That's what I thought at first, but it's only running with about 2msec ticker interval, or 500Hz.

jrobeson, 2015-10-12:
@andrewjfox : do the interrupts stop working altogether? or does it just fail with the timer part?

I'm having a similar problem with the RemoteIR library on two different NRF51822 boards, I see the interrupts still fire, just nothing happens.

andrewjfox, 2015-10-12:
Not sure if all interrupts stop, since the BLE core is still funcitonaing, however all of the interrupts attached to my application code stop. I have several inputs setup as InterruptIn and they stop responding. Dont know if the interrupts are firing, and code not running, or if interrupts stop firing.

Sometimes after about 20secs it all comes back to life, but most of the time I have to manually reset, which is a real pain since it's a BLE battery device and doesn't have an easy way of manually resetting, so I have to short the power pins.

jrobeson, 2015-10-12:
oh, i was gonna ask if you could say put a simple led toggle on there, but I guess that'd be a little difficult.

sander, 2015-10-26:
@andrewjfox, did you find a solution? I’m seeing similar delays and am also using multiple tickers and timeouts in my nRF51822 board.

andrewjfox, 2015-10-29:
No, no solution as yet. I've had to disable the PWM function and just have simple on/off control for LEDs until I can resolve the problem. I'm seeing timing inconsistencies more often now as I implement more features in the code, although I'm being really careful to not add more timers unless needed and keep their periods long to minimise number of interrupts. The issues all started once I included the bluetooth functions. My suspicion is that the softdevice uses a lot of system resources so that there's very little left for the rest of the device.

If you find a solution let me know

jrobeson, 2015-10-29:
so you've narrowed the problem to pwm?

jrobeson, 2015-10-29:
narrowed it is not right, since you're still having timing problems!  I do wonder though if there are pwm specific issues in interaction with the softdevice though.

jrobeson, 2015-10-29:
on that note, I'm not sure why the nordic sdk components in mbed don't include the nordic pwm api

andrewjfox, 2015-10-29:
Apologies for the confusion. I meant the software PWM I've implemented using the timers, not the mbed PWM objects.

jrobeson, 2015-10-29:
so why didn't you use the (mbed) native pwmout interface?

andrewjfox, 2015-10-29:
good question, 2 reasons,
1. Only 3 available and I'm using 1 for a speaker output, and the 2 RGB LEDs are driven seperately so need 6 PWMs (2 x Red, Green and Blue)
2. I'm trying to reduce power and discovered that the mbed Pwm objects can't be deleted (bug in api that doesn't delete deconstruct correctly). Even when not required the PWM timer is still running in background so uses power.
2b I tried driving speaker using timers, to be able to save power when not required, but as it turns out I have to use at least 1 Pwm object for the speaker output because the timers can't keep up to drive at reasonable audio frequency around 1kHz, so I'm not saving power anyway. But that's a separate problem

jrobeson, 2015-10-29:
I think they removed the 3 available restriction, but made them all share something (i can't recall which setting)

jrobeson, 2015-10-29:
can you file a bug about the your  number 2 issue?

jrobeson, 2015-11-09:
ping mbed maintainers?

andrewjfox, 2015-11-09:
created issue #1409 to cover PwmOut problem but haven't had any feedback as yet. Don't even know if anyone has looked at it.

0xc0170, 2015-11-09:
@andrewjfox There are some replies for pwmout dtor ;) Regarding this timer problem, can you create simple test-code snippet which we could run locally to reproduce ticker issue (or the snippet in your fist code is the simplest we can get?) ?

jrobeson, 2015-11-09:
I've been riding the coattails of this issue due to having similar problems using the RemoteIR library with a BLE Nano (16K ram) and the NRF51 DK.    Any of you folks ever used that one on any NRF51822 hardware?

andrewjfox, 2015-11-09:
I don't see the problem when the code is really simple and softdevice is not loaded. Any suggestions as to how to create simple test code that also simulates the same load on the cpu would be greatly appreciated.

crespum, 2016-02-09:
Has anyone had this problem again? I'm having problems with the tickers very similar to those described here an in #832 ([see my comment there](https://github.com/mbedmicro/mbed/issues/832#issuecomment-181480606)).

seajayshore, 2016-07-06:
Bumping this issue too. I seem to be seeing the same issue with multiple tickers stopping firing after a random duration ~3-7 mins when softDevice is running.

Any progress that people on this issue know of?

Also cross-posted [here]: (https://github.com/mbedmicro/mbed/issues/1533)

jrobeson, 2017-07-18:
Is this still an issue these days?

jrobeson, 2017-11-29:
is this still valid for nordic devices?

shirish47, 2018-04-04:
I am not sure but I have been facing a similar issue. I have a big project code so can't share it. The thing is every thing works perfectly for most of the time.
I am using mbed BLE with custom devices made based on redbear labs devices with same modules and programmed on arduino IDE. This works fine for short periods.

In normal case device sleeps for most of the time and a Ticker is used to maintain clock(for timestamping data) so every second the ticker updates a variable. 
Now there is another timer which is called at every 15 mins just to start advertising for alive( only advertises for 10 sec).

above code is the only part that should work without any external device trying to connect. when I leave this setup for a long time sometimes the device goes in a state that I am not sure of as no debug messages as it's not in loop() which usually gives a message that it woke up for a time update.

I have no other way to debug this as this happens only when running for like a day.
I am thinking of offsetting the other timer that is periodically called by some milliseconds so I think the above-mentioned problem can be avoided. Any more suggestion what state would it be in? how could I debug this?

Gecko05, 2018-04-13:
@shirish47 I have't really dived into multiple tickers or timers within an mbed BLE application but I'm currently trying to debug an issue with a ticker running around 10ms to increment a counter while updating data through the UART Ble Service, the thing is that when I use the mentioned ticker, the BLE Advertising and Service would stop working after 2 minutes of so.

0xc0170, 2018-07-02:
Hi, we updated tickers spec and improved testing for Mbed OS 5.9. Please create a new issue with updated information if this problem persists. 

For the latest 2 comments, it would be nice to have a code snippet to reproduce the issue on our side.

Closing this ticket.

closed 2018-07-02 by 0xc0170