ARMmbed/mbed-os #1382: Travis config - Migrate from legacy to container-based infrastructure

0xc0170, 2015-10-16:
As subject, port this to a new container based infrastructure. This message is shown for our last builds.

sg-, 2017-01-11:
@0xc0170 is this still something needed?

0xc0170, 2017-01-12:
Seems like we use ``sudo`` that is still not available in the container travis system. We are still running on the legacy one, see https://docs.travis-ci.com/user/migrating-from-legacy/ 

I assume this is minor at the moment, the only positive thing would be faster times.

cc @bridadan - if we are happy as it is, lets close this.

bridadan, 2017-01-12:
@0xc0170 I agree that this is minor. I think its probably ok to close as the Travis CI will most likely be phased out at some point?

closed 2017-10-27 by ChiefBureaucraticOfficer