ARMmbed/mbed-os #1403: STM platform RTC reset issue

star297, 2015-11-01:
Could we finally have a fix for the RTC reset issue.
Most of the STM's have a RTC-ISR register that can be checked if the RTC is running.

A simple check in the RTC_API.c does work, I have been using it for some time now:

if(RTC->ISR == 7) {     // RTC initialization and status register (RTC_ISR), cold start (with no backup domain power) RTC reset value 

Bypass the RTC initialise in this case that stops the register being reset.

I believe Martin did merge a PR from me a few weeks back for the F7 but that seems not to have happened.

0xc0170, 2015-11-02:
@bcostm @adustm

dbestm, 2015-11-13:
OK I take this in charge for all STM32 series.

dbestm, 2015-11-13:
We think it's better to use the flag INITS instead. To check if the RTC was initialized.
Is it fine for you?

star297, 2015-11-13:
Which ever you think will the best option, using RTC->ISR is mentioned in the manual as a way to check the state of the RTC, has been working as I mentioned above.

adustm, 2016-02-02:
Hello @star297 ,
Is it ok for you now ? Could we close this issue  ?
Cheers

star297, 2016-02-02:
@adustm 
@0xc0170 

No I'm afraid you can't.
I have loaded Mbed_dev rev: 58, it is not correct and will not work.

This part has to be changed/ included:

int rtc_isenabled(void)
{
    if ((RTC->ISR & RTC_ISR_INITS) ==  RTC_ISR_INITS){
        return 1;
    } else {
        return 0;
    }
}

All the "rtc_inited" code needs to be removed, it's no longer required and has no use here.

I have this implemented on several targets L476, L152, F4xx, F746 and is working correctly.

If you prefer I will send an individual PR with the working code for each of the targets I have fully tested.

Each target that uses the same embedded peripheral RTC in the Mbed line up will need this change to work properly. Some of the other 'low end' targets do not have the RTC->ISR register so must not be changed.

Paul

closed 2016-02-02 by star297

0xc0170, 2016-02-02:
@star297 please do, at least one, so we can test it