ARMmbed/mbed-os #1405: UARTs 4 and 5 on STM32 NUCLEOL152RE do not work

shaneleonard, 2015-11-02:
The following code, when run on the STM32 NUCLEO152RE, hangs indefinitely without outputting anything over the serial. This is the case for both UART 4 and UART 5. However, UARTs 1, 2, and 3 all work as expected. 

    #include "mbed.h"

    Serial uart4( PC_10, PC_11 );
    // Serial uart5( PC_12, PD_2  );
  
    int main() {
       uart4.baud(9600);
       // uart5.baud(9600);

       while(1) { 
            wait(1);
            uart4.printf("UART4\r\n");
            // uart5.printf("UART5\r\n");
       }
    }

This question forum has another similar example and more information: https://mbed.arm.com/questions/59656/Serial-What-are-the-Supported-pins/

shaneleonard, 2015-11-02:
The issue occurs in <code>libraries/mbed/targets/cmsis/TARGET_STM/TARGET_STM32L1/serial_api.c</code>. There is a typing error on several lines, where <code>#if defined(USART4_BASE)</code> is used instead of <code>#if defined(UART4_BASE)</code> (and similarly for UART 5). <code>UART4_BASE</code> and <code>UART5_BASE</code> are defined in <code>libraries/mbed/targets/cmsis/TARGET_STM/TARGET_STM32L1/TARGET_NUCLEO_L152RE/stm32l152xe.h</code>. 

Because <code>USART4_BASE</code> is not defined, the UART 4 clock tree is never initialized, and thus none of the serial functionality is enabled. <code>UART4_BASE</code> is the correct name, because UARTs 4 and 5 are actually UARTs and not USARTs like USART 1, 2, and 3.

In summary, changing <code>USART4_BASE</code> and <code>USART5_BASE</code> in <code>libraries/mbed/targets/cmsis/TARGET_STM/TARGET_STM32L1/serial_api.c</code> to <code>UART4_BASE</code> and <code>UART5_BASE</code> fixes the issue. This was verified using an STM32 NUCLEOL152RE and a logic analyzer.

I am submitting a pull request to close this issue.

shaneleonard, 2015-11-02:
Not sure if I should leave this issue open until the pull request is accepted, so I'm actually going to open this back up until the fix is integrated into master.

adustm, 2015-11-26:
Hello @shaneleonard,
I've seen your issue and I agree with the fix you've implemented. 
Do you confirm that the issue does not exist any more ?
Cheers

shaneleonard, 2015-11-26:
Yes, the issue is fixed. Sorry, I must have forgotten to close this ticket.

closed 2015-11-26 by shaneleonard