ARMmbed/mbed-os #1408: SystemCoreClock and USB clock settings dependency (STM32 series)

Willem23, 2015-11-03:
In case you want to use the USB device (or host) features the core clockspeed has to have certain values otherwise you can not get the correct USB clock of 48 MHz. On the F103 for example you can run at 72 MHz core speed with 8MHz external clocks (PLL: 8x9 = 72, USB: 72/1.5=48), but you can only get a maximum of 48 MHz core speed using the 8 MHz HSI (PLL: (8/2)x12 = 48, USB: 48/1=48). 
It would be nice to include #define settings for USB in system_stm32fxxx.c. It would be even better when these selections can be set without importing and modifying the mbed lib source.

See also discussion on issue 1096 regarding timeout for clock auto-select.

jeromecoutant, 2017-06-01:
@Willem23 please have a look on #4421 
Regards,

Willem23, 2017-06-02:
See comment for #4421

jeromecoutant, 2017-07-13:
Update for NUCLEO_F103RB is in #4740 
Clock source and USB configuration are now available in json files.

With default value : 
- HSE (at 72 MHz) is tried first, then HSI (at 64 MHz) in case of failure
- USB is available with HSE, but not with HSI

LMESTM, 2017-08-17:
#4740 has been merged - shall we close this issue ?

0xc0170, 2017-08-17:
@Willem23 reopen if still valid with new findings please

closed 2017-08-17 by 0xc0170