ARMmbed/mbed-os #1409: deleting PwmOut object crashes application

andrewjfox, 2015-11-04:
Hi, I originally posted this on mbed forum, but have also posted here.

I'm using the nRF51822-DK platform, and wanted to delete, or shutdown, PwmOut objects during sleep, as they use a bit of background power.

The app kept crashing so I've created a simple app to test, and it crashes after 2 cycles

```
/* testing power consumption 
    PwmOut dynamic allocation not working
*/
#include "mbed.h"

int main() {
    
    PwmOut *ptr;
    
    while(1) {
    
        ptr = new PwmOut(LED1);
        ptr->write(0.5);
        wait(1);
        ptr->write(1);
        wait(1);
        
        delete ptr;        
    }
}
```

LED flashes twice then program goes nuts, all LEDs blink randomly.  
I'm assuming this object supports dynamic allocation, since I've read in many other posts people wanted to delete objects to save power. I also think the object is not deleting correctly, and after a quick look in src, found that there's no desctructor for this object, but not sure if that's normal, and tbh am a little cautious about changing something after just a quick inspection. 

Looking forward to suggestions or comments

Thanks in advance
Andrew

PS, I've tried modifying mbed_src to include destructor in the api, and although it stops crashing, the Pwm timer is still runnning in background so no power saving

leibin2014, 2015-11-09:
How about call pwmout_free() in the desctructor?
The implement of pwmout_free() should be changed. I implemented the pwmout_free() as below:
void pwmout_free(pwmout_t* obj) {
   MBED_ASSERT(obj->pwm != (PWMName)NC);
    pwmout_write(obj, 0);
    PWM_taken[obj->pwm] = 0;
    timer_free();
    ppi_free(obj->pwm);    
    gpiote_free(obj->pin,obj->pwm);    
}

0xc0170, 2015-11-09:
@andrewjfox 
have you checked what is generated for implicitly-generated destructor, why is it crashing? 

There's pwmout_free function in the hal, which could be invoked as @leibin2014 pointed out, plus implementation of that function checked. There are platforms which have it empty for instance

andrewjfox, 2015-11-09:
I don't know why it's crashing, but given how simple the test is I suspect deleting the object is not freeing resources.

In the standard mbed library there is no destructor for PwmOut, however there is a pwmout_free function in the NRF51 target source. I've tried implementing a destructor for PwmOut and calling pwmout_free. 
```
~PwmOut() {
        pwmout_free(&_pwm);
    }
```
this stops the program crashing, but doesn't actually stop the Pwm from running in the background, so it is still using power. The pwmout_free function (pwmout_api.c in TARGET_NORDIC) only sets the pwm to 0, it doesn't disable it.
```
void pwmout_free(pwmout_t *obj)
{
    MBED_ASSERT(obj->pwm != (PWMName)NC);
    PWM_taken[obj->pwm] = 0;
    pwmout_write(obj, 0);
}

```

anpilog, 2015-11-10:
@andrewjfox: @leibin2014 gave you the solution to your problem.
We had same issue several months ago and resolved it in the way described few posts above.
```
void pwmout_free(pwmout_t* obj) {
  MBED_ASSERT(obj->pwm != (PWMName)NC);
  pwmout_write(obj, 0);
  PWM_taken[obj->pwm] = 0;
  timer_free();
  ppi_free(obj->pwm);

  gpiote_free(obj->pin,obj->pwm);
}
```
Regards.

andrewjfox, 2015-11-10:
Thanks all. 

The suggestion above appears to only stop PWM if all PWMs are freed, is this correct? Is it possible to stop a single PWM object? I've only started using the nRF51 and mbed so apologies if haven't understood how the low level works.

Does anyone know if this will be updated in official library release?

leibin2014, 2015-11-10:
@andrewjfox: All PWMs share the same timer. So the timer should be stopped after all PWMs are freed. Otherwise will effect the other PWMs which don't want to stop.

0xc0170, 2015-11-10:
@leibin2014 Will you send PR with this patch , you shared above?

andrewjfox, 2015-11-10:
Thanks to all for your help.

leibin2014, 2015-11-11:
@0xc0170 I have created the PR here:
https://github.com/mbedmicro/mbed/pull/1416

andrewjfox, 2015-11-19:
@leibin2014, #1416, I've implemented the changes as described and run some tests. I can't tell if the resource is being freed, but the power consumption is not dropping. I run the following sample code and instead of going to sleep, the loop keeps repeating. The Pwm timer must still be calling an interrupt that drops it out of sleep. If I remove the PwmOut object the code works as expected, going to sleep and current drops to 5uA. With the PwmOut object, the loop keeps repeating and current averages 4-5mA. Note The second for loop is simply to verify that code is dropping out of sleep().

```
#include "mbed.h"

DigitalOut test(LED1);

int main() {

    PwmOut *ptr;

    while(1) {

        ptr = new PwmOut(LED2);
        for(int i=0; i<5; i++) {
            test = 0;
            ptr->write(0);
            wait(1);
            test = 1;
            ptr->write(1);
            wait(1);
        }    
        delete ptr;        
        
        sleep();
        
        for(int i=0; i<3; i++) {
            test = 0;
            wait(0.2);
            test = 1;
            wait(0.2);
        }
        wait(1);
    }
}
```

Regards
Andrew

DerbselliconLabs, 2015-12-21:
@andrewjfox i just tried your test code with an additional call to sleep():  the current drops to 5uA.

#include "mbed.h"

DigitalOut test(LED1);

int main() {

    PwmOut *ptr;

    while(1) {

        ptr = new PwmOut(LED2);
        for(int i=0; i<5; i++) {
            test = 0;
            ptr->write(0);
            wait(1);
            test = 1;
            ptr->write(1);
            wait(1);
        }    
        delete ptr;        

        sleep();
        sleep();

        for(int i=0; i<3; i++) {
            test = 0;
            wait(0.2);
            test = 1;
            wait(0.2);
        }
        wait(1);
    }
}

zskdan, 2015-12-21:
@andrewjfox i just tried your test code with an additional call to sleep(): the current drops to 5uA.

#include "mbed.h"

DigitalOut test(LED1);

int main() {

PwmOut *ptr;

while(1) {

    ptr = new PwmOut(LED2);
    for(int i=0; i<5; i++) {
        test = 0;
        ptr->write(0);
        wait(1);
        test = 1;
        ptr->write(1);
        wait(1);
    }    
    delete ptr;        

    sleep();
    sleep();

    for(int i=0; i<3; i++) {
        test = 0;
        wait(0.2);
        test = 1;
        wait(0.2);
    }
    wait(1);
}
}

closed 2017-10-27 by ChiefBureaucraticOfficer