ARMmbed/mbed-os #1413: DISCO-L476VG Pinmap error: wrong port number

star297, 2015-11-07:
InterruptIn up(JOYSTICK_UP);

Creates this error. Also screws up interrupt and Ticker functions. Other button inputs work fine.
Had a quick look, but couldn't see anything in the pin defines.

0xc0170, 2015-11-11:
cc @bcostm

lcafaro, 2016-02-03:
Confirmed on DISCO-L476VG.

Seems like a bad definition of JOYSTICK_UP in PinNames.h?

bcostm, 2016-02-04:
I've just rechecked the board schematic and the pins definitions are OK. There is something else that causes this problem.
I have also rechecked the DISCO_L476VG_Leds_Buttons example and I confirm also the problem when the Up pin is configured. Without it it is OK.
I will analyse this deeper.

star297, 2016-02-18:
See #1509.
Setting the NVIC_RAM_VECTOR_ADDRESS to 0x10000000 instead of 0x20000000 in the cmsis_nvic.c file has fixed this problem.

closed 2016-02-18 by star297