ARMmbed/mbed-os #1419: default serial baud rate is incorrect on Nucleo-F070RB platform

galenseitz, 2015-11-11:
The default baud rate for Serial is documented as being 9600, but when the Nucleo_printf example is built for the Nucleo-F070RB platform using mbed revision 109, the actual baud rate is 57600.  It appears that this was broken by commit d571b16e907bd404dc5bbba7ae18ad79d5899b75.  Reviewing the commit, I suspect that this problem exists for all STMF0 platforms.  More debugging details can be found here: https://developer.mbed.org/questions/61404/baud-rate-problem-on-Nucleo-F070RB-platf/#answer8910.

FYI, I have also noticed that the default SPI bus clock frequency on the Nucleo-F070RB platform is 1.5 MHz rather than the documented 1 MHz default.  I haven't attempted to track down the source of this problem.

galen

0xc0170, 2015-11-12:
@adustm

adustm, 2015-11-12:
Hi, 
I have made the following trials without being able to reproduce your issue:
With online compiler, using the last official version of mbed library (version 109) and NUCLEO-F070RB platform : the printf works fine and I was also able to change the baudrate.
With github test environment, using the up-to-date mbedmicro:master code: I was able to compile with gcc-arm and use the printf.
The received baudrate is correct and not multiplied.

Coud you make sure that SB16 and SB50 (marked MC0) are soldered on the back side of you NUCLEO-F070RB board ? They are supposed to be closed by default.

galenseitz, 2015-11-12:
I have confirmed that SB16 and SB50 are bridged with a 0 ohm resistor.  My board is marked MB1136 C-03.

I reran the Nucleo_printf example and I am still seeing the problem.  Using the online compiler, I imported Nucleo_printf, selecting "Update all libraries to latest revision".  I then built and downloaded the example to my board.  I observed a baud rate of 57600, both on an oscilloscope(at SB13) and via a terminal emulator(kermit).  Next imported a new copy of Nucleo_printf, but this time I switched the mbed library revision to 108.  Again I observed a baud rate of 57600.  Finally I imported a new copy of Nucleo_printf, but this time I switched the mbed library revision to 107.  At this point the baud rate changed to 9600.

adustm, 2015-11-13:
Hi,
Issue eventually reproduced.
If you use printf("Hello World !\n"); it has a correct baudrate, and if you use a global variable Serial pc(SERIAL_TX, SERIAL_RX); and pc.printf("Hello World !\n"); it has the wrong baudrate.
I am working on the bug fix.
Cheers

adustm, 2015-11-17:
Hello @galenseitz,
I just submitted a pull request for this issue.
Cheers

commit 57650768 (2016-01-04): Fix #1419 : solve baud_rate issue

Issue was : stdio printf is ok, serial.printf is failing, in case it is
inialized as a global variable. It works on mbed version 107.
Fix: revert to version 107 for systemxxx.c, mbed_overrides.c +
cmsis_nvic.c.
Modify cmsis_nvic.c to use direct addressing instead of a pointer (this
pointer is not yet initialized when called in systemxx.c)
Call HAL_Init in mbed_overrides.c for every platforms.
commit a63257b0 (2016-01-04): Merge pull request #1432 from adustm/DEV_FIX_1419_baudrateissue

[TARGET_STMF0] Fix #1419 baudrateissue
closed 2015-12-03 by 0xc0170