ARMmbed/mbed-os #1435: Support for STM32F7-Disco with GNU tools for ARM Embeeded processors

rgkandur, 2015-11-18:
In the current release of mbed sdk, the mbed library is only available to use with Keil compiler for STM32F7-Disco board and no corresponding library available to use with GNU ARM tools.  When can I expect to have mbed library available with GNU ARM tools.  Please consider this request at highest priority.  Also I would like to have the library built with 'hard' floating point instead of 'softfp'.

ohagendorf, 2015-11-18:
Please see PR #1436

The library was already available for gcc_arm. Only one entry in target.py was missing. The PR doesn't change the fpu config.

rgkandur, 2015-11-19:
Thanks for the quick reply.  May I know, when can I expect mbed library for STM32F7-Disco compatible to GNU ARM tools available in mbed sdk. I would expect the STM32F7-Disco mbed library to be built with hard fpu instead of softfp.

ohagendorf, 2015-11-20:
The PR is accepted. This issue could be closed.

closed 2015-11-20 by 0xc0170