ARMmbed/mbed-os #1440: Pls. align InterruptIn

betzw, 2015-11-19:
Could you pls. align the implementation of `InterruptIn` to how it i done now in mbedOS?!?

0xc0170, 2015-11-19:
I'll look at it if it's here, I recall call() method in function pointer does the check if it's not null, thus won't fail.

0xc0170, 2015-11-19:
Please provide more descriptive information - file reference, code snippet, etc

betzw, 2015-11-20:
The fix was [this](https://github.com/0xc0170/mbed-drivers/commit/2c0fe0426c9c037c120b17dc206738fadfbf76bc).
The story can be found [here](https://github.com/ARMmbed/mbed-drivers/pull/120).

betzw, 2015-11-25:
Pls. note that PR https://github.com/mbedmicro/mbed/pull/1443 does not fix completely the issue, as the checks before calling eventually a null pointer, see:
https://github.com/ARMmbed/mbed-drivers/blob/master/source/InterruptIn.cpp#L66
and
https://github.com/ARMmbed/mbed-drivers/blob/master/source/InterruptIn.cpp#L71
are missing!

PR #1443 (2015-11-26): InterrupIn - attach with null clears the function pointer
If we attach a function, then detach, a function pointer keeps pointing to an old function.

Fixes #1440
closed 2015-11-26 by 0xc0170