ARMmbed/mbed-os #1447: LPC11U35 Compiler/linker issue

neilt6, 2015-11-23:
Some of my previously working LPC11U35 programs are hard faulting before they even get to `main()`, even known-good versions with older libraries. In an isolated test, I was able to produce an even weirder problem in which the presence of a destructor in a class breaks USBDevice, while removing it "fixes" it. I think this may be a compiler/linker issue, and that it may affect other targets as well. More details here: https://developer.mbed.org/forum/bugs-suggestions/topic/17204/

neilt6, 2015-11-25:
Additional experiments have shown that the LPC11U68 and LPC1549 are also affected by this issue.

adamgreen, 2015-11-26:
I take it from your forum post that this is just happening with the online compiler?  If I take your **weird bug** sample and build it for my LPC11U24 board, it should reproduce the hard fault?

adamgreen, 2015-11-26:
**Note:**  I built your LPC11U35_WeirdBug sample for the LPC11U24 with the online compiler and it didn't fault.  When I break in with GDB, it is always just running through the main loop.

sg-, 2015-11-26:
I built online for the lpc1768 and was not able to reproduce either. Will keep trying different boards. I did notice that with both the connect false and the constructor false the hid device did enumerate which I found odd.

neilt6, 2015-11-26:
The example I posted doesn't hard fault, but USBDevice will break when the `DummyClass`'s destructor is present. Windows will report the USB device has "malfunctioned", but the loop will still run IIRC. Removing the destructor, or overriding `__aeabi_atexit()` fixes the problem. Compiling it offline using uVision 4 works too.

neilt6, 2015-11-26:
It's behaving like a memory corruption issue, different programs are being affected in different ways. While playing around with library versions in one of my projects, I was able to get a binary that didn't hard fault, but USBDevice wouldn't enumerate. Now that same combination is hard faulting...

adamgreen, 2015-11-26:
I see.  Too hard for me to debug without symbols then.  I will bow out and let you and Sam take care of it.

neilt6, 2015-11-26:
I think it might be a heap-related issue with the ARM micro toolchain, since overriding `__aeabi_atexit()` fixes it. http://www.keil.com/forum/21859/using-destructor-without-heap-results-in-stop-on-bkpt/

neilt6, 2015-11-26:
I just tested the LPC1768, and can confirm that it is NOT affected by this issue.

neilt6, 2015-11-26:
Here's something interesting, [startup_LPC11xx.S](https://github.com/mbedmicro/mbed/blob/9b7d23d47153c298a6d24de9a415202705889d11/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC11UXX/TOOLCHAIN_ARM_MICRO/TARGET_LPC11U35_401/startup_LPC11xx.S) doesn't appear to be defining `__heap_base` and `__heap_limit` properly. The assumption seems to be that `__user_setup_stackheap()` in [sys.cpp](https://github.com/mbedmicro/mbed/blob/9b7d23d47153c298a6d24de9a415202705889d11/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC11UXX/TOOLCHAIN_ARM_MICRO/sys.cpp) will initialize the heap. However, according to [Keil's documentation](http://www.keil.com/support/man/docs/armlib/armlib_chr1358938929242.htm), that method isn't applicable to microlib... I would try changing the values, but unfortunately the online compiler isn't letting me edit startup_LPC11xx.S.

closed 2017-10-27 by ChiefBureaucraticOfficer