ARMmbed/mbed-os #1455: MBED Version 110 Does not have the GCC Toolchain changes submitted by ohagendorf as per PR1438

sandeepach, 2015-11-26:
MBED Version 110 Does not have the GCC Toolchain changes submitted by ohagendorf as per PR1438. 
Please merge it to the release. I understand from Martin Kojtal's reply on MBED group that build_release.py needs to be modified for this.

sandeepach, 2015-11-26:
This is for STM32F746 Discovery board.

0xc0170, 2015-11-26:
cc @bcostm

ohagendorf, 2015-11-26:
Hello @sandeepach 

I just checked the current master version and everything looks fine for me. My changes are included.

Could you please explain, what exactly is your problem.

Olaf

0xc0170, 2015-11-26:
@ohagendorf gcc support is not part of the release as it's not defined here : https://github.com/mbedmicro/mbed/blob/master/workspace_tools/build_release.py#L89

sandeepach, 2015-11-26:
Hello Olaf
I think Martin is right, my problem is https://developer.mbed.org/users/mbed_official/code/mbed/ latest version(110) does not have the object files built with GCC for STM32F746 Discovery. 
This was my original problem which was said fixed as per PR1438. 
Sandeep.

ohagendorf, 2015-11-26:
Hello @sandeepach 

please see PR #1456.

Regards
Olaf

closed 2015-11-27 by 0xc0170