ARMmbed/mbed-os #1465: LPC1768 Repeated Timer::read_ms() discontinuity in returned value

gbeardall, 2015-12-02:
If **Timer::read_ms()** is called repeatedly then at apprx (2^32)/2000 milli-secs (about 36 mins) the timer value jumps to (2^32)*1999/2000 and then continues from there, and wraps as normal. Each time it reaches the 1st value above, the jump occurs. Additionally, __read_us()__ doesn’t seem to suffer from the same problem.

    void main() {

        Timer timer1;

        uint32_t timer_old = 0;

        timer1.reset();
        timer1.start();

        timer_old = timer1.read_ms(); // init

        while(1) {

            // Calculate timer difference since last iteration
            uint32_t timer_now = timer1.read_ms();
            uint32_t timer_delta  = timer_now - timer_old;

            pc.printf("%u %u\n", timer_now, timer_delta);

            // Check timer delta not too big
            if(timer_delta > 500u) {
                // Should never get here as loop time should always be less than 500ms

                pc.printf("*** delta error now=%u old=%u delta=%u ratio=%.2f\n", 
                    timer_now, timer_old, timer_delta, 
                    ((float)timer_now)/timer_old);
                
                while(pc.readable()) pc.getc(); pc.getc(); // flush, then pause on error
            } // if
             
            timer_old = timer_now; // update old

        } // while

    } // main

adamgreen, 2015-12-07:
This seems like this behavior is to be expected.  This is caused by the `/ 1000` [here](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/common/Timer.cpp#L51).  The 1000 in the divisor accounts for part of the 2000 roll-over factor that you are seeing.  The other part is due to the 1 bit lost by using signed division instead of unsigned.

If you want your timer value to roll-over without the weird wrap then you should use read_us() instead.  My recommendation would be to just document this issue and leave the SDK code as is.

0xc0170, 2015-12-07:
> The other part is due to the 1 bit lost by using signed division instead of unsigned.

:tired_face: 

> If you want your timer value to roll-over without the weird wrap then you should use read_us() instead. My recommendation would be to just document this issue and leave the SDK code as is.

We should definitely add a comment to API

gbeardall, 2015-12-08:
Ok, my misconception from the documentation was that read_ms() was a synthesised full 32 bit milli-sec counter. It might be worth documenting that as well as the weird wrap problem.

closed 2017-10-27 by ChiefBureaucraticOfficer