ARMmbed/mbed-os #1466: Nucleo L152RE not waking from Deep Sleep

star297, 2015-12-04:
L152RE will wake from Sleep but not Deep Sleep I'm sure this was working a few months ago the last time I tried.

0xc0170, 2015-12-08:
cc @bcostm @dbestm

star297, 2016-01-20:
Has this been fixed yet?

0xc0170, 2016-01-21:
I don't recall this has been fixed yet.

bcostm, 2016-01-21:
I don't know. Need to recheck with latest mbed library and advise what should be done.

star297, 2016-01-24:
Further checks, I found it did wake up using HSI.
Using external clock, it is set to start on EXTC first then XTAL.
My board, as many will be, does not have the MCO link soldered as default.

The fix is to transpose the EXTC and XTAL order starting at line 455 in the system_stm32l1xx.c file as below:

void SetSysClock(void)
{
  /* 1- Try to start with HSE and external xtal */
#if USE_PLL_HSE_XTAL != 0
  if (SetSysClock_PLL_HSE(0) == 0)
#endif
  {
    /* 2- If fail try to start with HSE and external clock */
    #if USE_PLL_HSE_EXTC != 0
    if (SetSysClock_PLL_HSE(1) == 0)
    #endif

This should now try to start in the following order, XTAL then EXTC and finally HSI.
Working on my board with both 8MHz & 32KHz crystals fitted.
Needs to be tested on a board with MCO link soldered.
If that works then this change can be made and #1466 can be closed.

bcostm, 2016-01-27:
Just checked the Nucleo sleep example on Nucleo_L152RE. The example is OK with mbed v112 and mbed-dev v53.

Concerning the change in the clock configuration, this has to be aligned with the default h/w clock configuration (i.e. MCO connected or not with SB16/SB50 solder bridges) on the board.
Following the last Nucleo board schematic, all the nucleo boards (excepted the L476RG) are clocked by default using the MCO coming from the STlink part.
So, for the L152RE we must keep the actual order: 1) try with MCO, 2) try with XTAL, 3) then HSI.

mfiore02, 2016-06-21:
@star297 what are your wake method(s)? We recently encountered a similar issue on the Nucleo L152 board. When we configured the device for RTC Alarm wakeup and GPIO wakeup using the wake pin connected to the user button, the device wouldn't wake up from sleep because the biasing on the user button was backwards - pull up with press connecting to ground instead of pull down with press connecting to 3.3V. RTC wakeup alone was fine. Anyway, holding the user button down while resetting the device and then letting go when we wanted to wake up took care of the issue. Our custom HW will have the correct biasing to avoid this, but I thought it may help you.

The [errata document](http://www.st.com/content/ccc/resource/technical/document/errata_sheet/83/ae/21/a3/1c/a4/44/44/DM00104204.pdf/files/DM00104204.pdf/jcr:content/translations/en.DM00104204.pdf) for the nucleo boards mentions this a little bit.

sg-, 2017-01-16:
@star297 Has this been resolved?

jeromecoutant, 2017-05-22:
Maybe we close this issue ?

closed 2017-05-22 by 0xc0170