ARMmbed/mbed-os #1475: STM32F7 HardFP Support

sandeepach, 2015-12-14:
I think STM32F746 Discovery board supports HARD FP. 
However the mbed library uses only softfp. 
I think to best use the capabilities of the board, HardFP should be supported.

0xc0170, 2015-12-14:
This is related #1467.  The target is currently defined as M7 without fpu ;)

sandeepach, 2016-01-11:
Any updates on this? 
I got the MBED Library version to work when the rest of the C source code is also compiled without FPU. 
However, this is working sub-optimally. Any plans to merge this PR? I see it is idle for too long.

0xc0170, 2016-01-11:
I am looking at it, got some problems when testing, therefore need more time. I'll update it today with the progress

0xc0170, 2016-01-21:
We merged the fpu changes, please test and report if any problems.

closed 2016-01-21 by 0xc0170