ARMmbed/mbed-os #1480: STM32F4 RTC broken after soft reset - mbed lib rev 111

mfiore02, 2015-12-22:
[This commit](https://github.com/mbedmicro/mbed/commit/feac60b519675a78423bb7ac1f71f790cc5f5e4a) added a check of the RTC_ISR_INITS flag in the RTC_ISR register in rtc_init(). This change has broken some RTC functionality for at least the STM32F411 processor.

The following code demonstrates the issue. The RTC is set and read properly when the device is cold booted, but a constant value, 946684800, is read from the RTC after a soft reset of the device.

```c++
#include "mbed.h"
 
int main() {
    Serial usb(USBTX, USBRX);
    usb.baud(115200);
    usb.printf("starting\r\n");
    
    set_time(1450813642);
    
    while (true) {
        time_t seconds = time(NULL);
        usb.printf("time: %d\r\n", seconds);
        wait(5);
    }
    
    return 0;
}
```

mfiore02, 2015-12-23:
I added some code to print the RTC registers before and after the set_time call.

```c++
#include "mbed.h"
 
int main() {
    DigitalIn pb(USER_BUTTON);
    Serial usb(USBTX, USBRX);
    usb.baud(115200);
    
    while (pb == 1);
    usb.printf("starting\r\n");

    usb.printf("RTC regs before set_time()\r\n");
    usb.printf("RTC TR:       %#X\r\n", RTC->TR);
    usb.printf("RTC DR:       %#X\r\n", RTC->DR);
    usb.printf("RTC CR:       %#X\r\n", RTC->CR);
    usb.printf("RTC ISR:      %#X\r\n", RTC->ISR);
    usb.printf("RTC PRER:     %#X\r\n", RTC->PRER);
    usb.printf("RTC WUTR:     %#X\r\n", RTC->WUTR);
    usb.printf("RTC CALIBR:   %#X\r\n", RTC->CALIBR);
    usb.printf("RTC ALRMAR:   %#X\r\n", RTC->ALRMAR);
    usb.printf("RTC ALRMBR:   %#X\r\n", RTC->ALRMBR);
    usb.printf("RTC WPR:      %#X\r\n", RTC->WPR);
    usb.printf("RTC SSR:      %#X\r\n", RTC->SSR);
    usb.printf("RTC SHIFTR:   %#X\r\n", RTC->SHIFTR);
    usb.printf("RTC TSTR:     %#X\r\n", RTC->TSTR);
    usb.printf("RTC TSDR:     %#X\r\n", RTC->TSDR);
    usb.printf("RTC TSSSR:    %#X\r\n", RTC->TSSSR);
    usb.printf("RTC CALR:     %#X\r\n", RTC->CALR);
    usb.printf("RTC TAFCR:    %#X\r\n", RTC->TAFCR);
    usb.printf("RTC ALRMASSR: %#X\r\n", RTC->ALRMASSR);
    usb.printf("RTC ALRMBSSR: %#X\r\n", RTC->ALRMBSSR);
    
    set_time(1450813642);
    
    usb.printf("RTC regs after set_time()\r\n");
    usb.printf("RTC TR:       %#X\r\n", RTC->TR);
    usb.printf("RTC DR:       %#X\r\n", RTC->DR);
    usb.printf("RTC CR:       %#X\r\n", RTC->CR);
    usb.printf("RTC ISR:      %#X\r\n", RTC->ISR);
    usb.printf("RTC PRER:     %#X\r\n", RTC->PRER);
    usb.printf("RTC WUTR:     %#X\r\n", RTC->WUTR);
    usb.printf("RTC CALIBR:   %#X\r\n", RTC->CALIBR);
    usb.printf("RTC ALRMAR:   %#X\r\n", RTC->ALRMAR);
    usb.printf("RTC ALRMBR:   %#X\r\n", RTC->ALRMBR);
    usb.printf("RTC WPR:      %#X\r\n", RTC->WPR);
    usb.printf("RTC SSR:      %#X\r\n", RTC->SSR);
    usb.printf("RTC SHIFTR:   %#X\r\n", RTC->SHIFTR);
    usb.printf("RTC TSTR:     %#X\r\n", RTC->TSTR);
    usb.printf("RTC TSDR:     %#X\r\n", RTC->TSDR);
    usb.printf("RTC TSSSR:    %#X\r\n", RTC->TSSSR);
    usb.printf("RTC CALR:     %#X\r\n", RTC->CALR);
    usb.printf("RTC TAFCR:    %#X\r\n", RTC->TAFCR);
    usb.printf("RTC ALRMASSR: %#X\r\n", RTC->ALRMASSR);
    usb.printf("RTC ALRMBSSR: %#X\r\n", RTC->ALRMBSSR);
    
    while (true) {
        time_t seconds = time(NULL);
        usb.printf("time: %d\r\n", seconds);
        wait(5);
    }
    
    return 0;
}
```

The results are below. The file "rtc_good.txt" is the results from a cold boot. The file "rtc_bad.txt" is from the soft reset after the boot.

[rtc_bad.txt](https://github.com/mbedmicro/mbed/files/70978/rtc_bad.txt)
[rtc_good.txt](https://github.com/mbedmicro/mbed/files/70979/rtc_good.txt)

I should also note that the time(NULL) calls fail even if set_time() isn't called after a soft reset.

0xc0170, 2015-12-23:
cc @dbestm @bcostm 

Lets fix this for the next release.

From that commit referenced above, this line breaks it ``` if ((RTC->ISR & RTC_ISR_INITS) ==  RTC_ISR_INITS) return; ```

mfiore02, 2015-12-29:
If the intended behavior could be preserved, that would be valuable. It is useful to be able to keep the RTC running across soft resets or sleep modes and still use the mbed rtc api.

commit 26e3c619 (2016-01-12): [STM all] Fix #1480: change location of check for RTC already initialised

Need to keep PWR_CLK_ENABLE and LSE LSI oscillator configuration.
Just skip the HAL_RTC_Init in case INITS flag is already set.
closed 2016-01-12 by 0xc0170