ARMmbed/mbed-os #1486: Exporting to Atmel Studio

sg-, 2015-12-31:
```
$python project.py -m SAMD21G18A -i atmelstudio -p 27 -c
```
Using AS7 and the update wizard there is a problem with the generated projects if you click the 'migrate to current infrastructure' box. It adds an additional Device Startup folder causing duplicate symbols during linking

Appears to works OK with this box unchecked.

0xc0170, 2016-01-28:
@kpurusho Please have a look

kpurusho, 2016-02-02:
For projects which contains self contained CMSIS and startup related file (which is the case for mbed exported projects), it is required to un-check 'migrate to current infrastructure' box.

0xc0170, 2016-02-02:
This might be described in the GettingStarted.htm ? It translates into ``` <meta http-equiv="refresh" content="0; url=http://mbed.org/handbook/Getting-Started-mbed-Exporters#atmelstudio"/>```

sg-, 2016-02-16:
https://developer.mbed.org/handbook/Getting-Started-mbed-Exporters#atmelstudio

Good idea. Next time I use atmel studio I'll capture a screen shot unless someone has one and wants to email me or upload and share the URL.

closed 2016-02-16 by sg-