ARMmbed/mbed-os #1494: Simple program of NUCLEO_F030R8 dose not work.

dinau, 2016-01-07:
Latest code of github version for Nucleo-F030R8  is probably broken.
Target compiler is armcc.
Below code dose not work.
```
#include "mbed.h"
DigitalOut led1(LED1);
int i;
int main() {
    while (1) {
        led1 = 1;
        wait_ms( 500);
        led1 = 0;
        wait_ms( 500 );
        printf("\n %03d: Rev.%d: Clock=%d Hz",i++,MBED_LIBRARY_VERSION,SystemCoreClock);
    }
}
```
The code released as rev.111  is ok.

dinau

0xc0170, 2016-01-07:
Does not work means? Can you specify more details, what does it print, the led interval is not correct, print does not work, the systemcoreclock is wrong, or?

dinau, 2016-01-07:
Hello,
(1)  LED dose not blink once. (state off)
(2) Nothing is printed.
 I can't  even do  "printf debug" ,therefore I don't konw any more details at this moment.

dinau

0xc0170, 2016-01-07:
Is this the one causing this : https://github.com/mbedmicro/mbed/pull/1432 ? Can you try to revert that patch to see if it works? In the HAL , the only PR within the time frame was from you.

dinau, 2016-01-07:
I've reverted code below (before merge #1432).
"LPC4337 stdio fixes for lpcxpresso board"
( GIT SHA : 7899cff5f676742c244cca2908aea850227f6b53)

Everything is ok.  

dinau.

dinau, 2016-01-07:
I think the cause is probably #1432.

dinau

adustm, 2016-01-08:
Hello,
I'll have a look

adustm, 2016-01-08:
Hello,
I fixed the issue #1494 in PR #1496.
The SYSCFG is not clocked, so the code : SYSCFG->CFGR1 |= SYSCFG_CFGR1_MEM_MODE was not applied.
In F030RC the function NVIC_SetVector is called twice in the HAL_InitTick function. 
    NVIC_SetVector(TIM_MST_UP_IRQ, (uint32_t)timer_update_irq_handler);
    NVIC_EnableIRQ(TIM_MST_UP_IRQ);
    NVIC_SetVector(TIM_MST_OC_IRQ, (uint32_t)timer_oc_irq_handler);
    NVIC_EnableIRQ(TIM_MST_OC_IRQ);

As SYSCFG->CFGR1 is still equal to 0 (because not clocked), the 2nd call to this function was then overwritting the address of the interrupt TIM_MST_UP_IRQ. 
At the next occurence of this interrupt, CPU does not know where to branch anymore, and is block in the interrupt handler.

I used the test MBED_25 to validate the fix, as it is quite similar to the main used in this issue.
Test MBED_25 was passed on every F0 targets. It used to crash before the fix for NUCLEO_F030R8 and NUCLEO_F070RB.
Now it passes "OK" on every F0 platforms : NUCLEO_F030R8, NUCLEO_F031K6, NUCLEO_F042K6, NUCLEO_F070RB, NUCLEO_F072RB, NUCLEO_F091RC.

dinau, 2016-01-09:
Hello,
I've confirmed that the simple program I showed above works well with
#1494.

Thank you.

dinau

closed 2016-01-11 by 0xc0170