ARMmbed/mbed-os #1509: STM32L4 InterruptIn has stopped working in Mbed-dev rev:58 Mbed rev: 112

star297, 2016-01-20:
The title says it all.
The example:
https://developer.mbed.org/teams/ST/code/Nucleo_read_button_interrupt/?platform=ST-Nucleo-L476RG
no longer works, program freezes when button is pressed.
Noticed this on Mbed-dev rev:58 Mbed rev: 112, but may have broke before in previous revisions.

star297, 2016-01-23:
Not so sure it is to do with revisions.
Something I found, if I simply add AnalogIn aIn(PA_1);
It runs correctly.
Why we need to set an AnalogIn pin? 
How does this affect the InertupIn function?

0xc0170, 2016-01-25:
cc @bcostm @adustm

star297, 2016-01-25:
I have noticed that the L476 does not require the LowPowerConfiguration code to achieve low power in deepsleep, where similar code is needed for the L152. It appears the GPIO's initialise in the low power mode. Perhaps the problem is with the GPIO's latching the IRQ's and these need to be configured in the same way as the L152 on start up, then applying LowPowerConfiguration code to set the unused GPIO's in the user program.

bcostm, 2016-01-27:
I've just checked the Nucleo_sleep example on mbed IDE with mbed v112 and mbed-dev v53 and I confirm that the pressed function is nvever entered when the user_button is pressed.

I've exported this example on uVision using mbed-dev and it is OK with sleep and deepsleep.
I've also checked using ARM_MICRO and ARM_STD lib and -O0 and -O3 options. In all cases it's OK.

So, it seems there is something wrong only with the mbed IDE but I have no idea what?

star297, 2016-01-27:
This example shows the problem:
https://developer.mbed.org/teams/ST/code/Nucleo_read_button_interrupt/?platform=ST-Nucleo-L476RG

I tried using uVision and was not working.
However if you change the pin define order to this:
 
DigitalOut myled(LED1);
InterruptIn mybutton(USER_BUTTON);

then it does work on Mbed and uVision (without defining the AnalogIn pin).

However before we go any further the Mbed IDE/Compiler need setting correctly.
Build Details is only showing 512K flash and 32k RAM. It should be 1M Flash and 128K RAM.
This is also true with the Disco-L476 and is suffering from similar issues.
Also issue #1348 needs checking.
I think all these problems are tied up and could simply be down to the complier WEB interface.

@0xc0170 Martin could you get the WEB guys to check and correct these settings for both Targets please, we can't move on until this is fixed.

star297, 2016-02-15:
cc @bcostm @adustm 

Guys, could you take a good look at the interrupt functionality as soon as possible please.
The Mbed Flash/RAM settings are now correct.
It appears there are other problems emerging, MBED-RTOS is also not running.
It is not revision dependant, not been working properly from the outset.
Setting multiple interrupts causes all sorts of lockup and weird code jumping problems.
In some cases enabling one defined interrupt pin causes all the other defined interrupt pins to become active.

bcostm, 2016-02-17:
Problem confirmed with Nucleo_read_button_interrupt example + mbed v114 or mbed-dev v65 + mbedIDE + NUCLEO_L476 board. But same example works on the DISCO_L476 board.

I made a diff of DISCO and NUCLEO cmsis folders and there is a difference in the stm32l476xx.icf files. SRAM1/SRAM2 regions are reverted. It could be due to this but it needs to be verified.

I'll do it tomorrow.

adamgreen, 2016-02-17:
If you can attach GDB to the board when the online built code is failing, you should still be able to grab some information about the state of the device during the failure.  You just won't have symbols so it will make things a bit harder.

If you do happen to do this and it breaks in successfully, you can grab a snapshot of the system and post a link to it for me to look at.  These GDB commands will cause it to dump the important state of the device:
```
set pagination off
set logging on
set var $ptr=0x10000000
while $ptr < 0x10008000
x/4wx $ptr
set var $ptr+=16
end
info all-registers
set logging off
set pagination on
```
I would need access to the GDB.TXT created by the above commands and the mbed produced .bin file which was running on the device.  With those 2 files I can load up the state of the device under GDB remotely and take a look.

bcostm, 2016-02-18:
We've discovered that NVIC_RAM_VECTOR_ADDRESS is set to 0x20000000 instead of 0x10000000 in the msis_nvic.c file.
I checked the 0x10000000 value in the mbed IDE with mbed-dev and the Nucleo_read_button_interrupt example works with InterruptIn placed before or after DigitalOut. Can you please try also ? If ok for you too I will send a pull request.

star297, 2016-02-18:
Spot on, that has fixed the interrupt problem.
Tried it with multiple interrupts triggering Tickers, works stable as expected.
Also checked using HSE @ 80MHz. 
With the RTC now working, the L476 is pretty much sorted :+1:

bcostm, 2016-02-19:
ok cool :smiley: thanks for verifying it.
I will send a PR asap.

sg-, 2017-01-16:
Confirmed to be resolved.

closed 2017-01-16 by sg-