ARMmbed/mbed-os #1510: PwmOut broken on LPC11U68, LPC1549, possibly others

neilt6, 2016-01-20:
PwmOut is currently broken on the LPC11U68, LPC1549, and possibly other LPC targets using SCTs for PWM output. The period and pulsewidth are both off by 1, and the input clock is being scaled to 1MHz making high frequency output impossible. I'm currently investigating fixes for these issues.

neilt6, 2016-01-20:
Mystery 1 solved: the match reload registers are -1 encoded. Therefore, the correct MATCHREL0 value for a 4μs period is 3, not 4. However, this means that a 0μs pulse width isn't possible, since 0 = 1μs. I'm currently investigating a workaround.

0xc0170, 2016-01-21:
Thanks for looking at this, please reference this issue in the PR with a fix

neilt6, 2016-01-21:
I've submit my pull request for review. The off-by-one errors have been fixed, and the counter clock now runs at the system clock speed, which is consistent with other platforms like the LPC11U24. In addition, the counter now halts when the pulse-width is 0μs in order to produce true 0% duty cycle output.

toyowata, 2016-04-11:
@neilt6 

SCT based PwmOut issue was fixed for both LPC824 and LPC812 as well.  I believe we fixed all existing SCT based HAL implementation and this issue can closed.

neilt6, 2016-04-11:
Thanks @toyowata

closed 2016-04-11 by neilt6