ARMmbed/mbed-os #1511: coupling between RTOS and STM32F4 stop mode

mfiore02, 2016-01-20:
I have been testing stop mode on the STM32F4 processors for an application. I've noticed a couple issues:

1) When using the online compiler, I need to include the mbed-rtos library in my project in order for the device to ever wake up from stop mode. If I remove the rtos library from my project and rebuild, my device never wakes up. The application is using mbed library revision 112 and mbed-rtos revision 101. All beta modes are turned off. I've reproduced this issue with both F411 Nucleo and F401 Nucleo boards. My application can be found [here](https://developer.mbed.org/users/mfiore/code/stm32f4-stop-mode/).

2) When building this same application offline using the mbed sdk (mbed_lib_rev112 tag) and the arm-none-eabi toolchain (version 4.9.3.2015q3-1trusty1 on 64 bit Ubuntu 14.04), I need to go even further to get the application to function properly. Simply invoking the make script with the --rtos option isn't enough; I have to actually include the rtos header and put a thread into my application for the device to properly wake up from stop mode. The following modified version of the original application works when the --rtos option is used with the make script. This was again verified on both the F411 Nucleo and F401 Nucleo boards.

```c++
#include "mbed.h"
#include "rtos.h"
#include <time.h>
#include <rtc_api.h>

DigitalOut led(LED1);

void thread_func(const void*) {
    while (true) {
        led = !led;
        osDelay(200);
    }
}

int is_rtc_enabled(void) {
    return (RTC ->ISR & RTC_ISR_INITS ) && (RTC ->ISR & RTC_ISR_RSF );
}

int main() {
    // attach our custom RTC enabled function so time can survive across sleeps & resets
    attach_rtc(&rtc_read, &rtc_write, &rtc_init, &is_rtc_enabled);
    if (!is_rtc_enabled()) 
        set_time(0);
    RTC_HandleTypeDef RtcHandle;
    RtcHandle.Instance = RTC;

    Thread t(thread_func);

    int interval = 4;
            
    while(true) {
        
        __PWR_CLK_ENABLE();
        HAL_PWR_EnableBkUpAccess();
        
        HAL_PWREx_EnableMainRegulatorLowVoltage();
        HAL_PWREx_EnableFlashPowerDown();
        HAL_PWREx_EnableLowRegulatorLowVoltage();
    
        // make sure RTC and LSE stay on while we're asleep
        RCC ->BDCR |= RCC_BDCR_LSEON;
        RCC ->BDCR |= RCC_BDCR_RTCEN;

        printf("configuring wakeup timer for %lu seconds from now\r\n", interval);

        HAL_RTCEx_SetWakeUpTimer_IT(&RtcHandle, interval, RTC_WAKEUPCLOCK_CK_SPRE_16BITS );

        printf("entering sleep mode %08x\r\n", RTC->ISR);
        fflush(stdout);
        
        // clear any pending interrupts
        EXTI ->PR = (1 << 22);

        HAL_PWR_EnterSTOPMode(PWR_LOWPOWERREGULATOR_ON, PWR_STOPENTRY_WFI );
                
        HAL_PWREx_DisableMainRegulatorLowVoltage();
        HAL_PWREx_DisableFlashPowerDown();
        HAL_PWREx_DisableLowRegulatorLowVoltage();
        
        __HAL_RTC_WAKEUPTIMER_CLEAR_FLAG(&RtcHandle, RTC_FLAG_WUTF);

        // reconfigure PLL
        SetSysClock();
        wait(1.0);
        printf("wake\r\n");    
    }

    return 0;    
}
```

I noticed that the online compiler seems to be building the rtos library from source (and linking against it), while the offline mbed sdk links against the rtos archive generated when the build script is run. Is this difference be part of the issue?

I noticed that the rtos library has additional startup code for the each toolchain. Does this replace the startup code in the standard mbed library or are they both used? Is there some critical piece of startup code that is only getting executed when the rtos library is involved?

I would like to have the following questions answered/issues resolved.

1) Why does stop mode on the STM32F4 need mbed-rtos to function properly?
2) Why is the behavior between the online compiler and the offline mbed sdk so different?

0xc0170, 2016-01-21:
cc @bcostm @adustm

bcostm, 2016-01-21:
1) It has never be the case. Need to recheck with latest mbed library and advise what should be done.
2) We (as ST) cannot do anything on the mbed IDE

mfiore02, 2016-01-21:
I tested the first case (online compiler) again using the latest revision of the mbed-dev library (rev 53) instead of the precompiled mbed library. The results were the same.

I will test with an older version of the mbed library as well.

@bcostm, I understand that you can't modify the mbed IDE, but my understanding is that it is using the mbed sdk with the ARM/Keil toolchain under the hood. @0xc0170, is that correct?

mfiore02, 2016-01-21:
I also tested the first case with revs 110 and 104 of the mbed library. The results were the same, so it seems that this behavior has been around for quite a while.

0xc0170, 2016-01-22:
> @bcostm, I understand that you can't modify the mbed IDE, but my understanding is that it is using the mbed sdk with the ARM/Keil toolchain under the hood. @0xc0170, is that correct?

Yes, to get the same use ARMCC (v5.06 recommended). Usually when you can't reproduce locally, use ARM or uARM (default toolchain for target), no debug-info (let the optimization be the same). You should get the same result Use project.py to export a project to get better understanding whats going on.

mfiore02, 2016-01-26:
This could be a showstopping issue for some of our customers (if they don't want to use RTOS).

Has anybody at ARM or STM  been able to reproduce this issue? Is anyone looking into it?

@sg- any thoughts?

0xc0170, 2016-01-26:
I'll have a look tomorrow, the code snippet is using specific ST HAL. Did you explore what is going on, why RTOS changes it, does it wake up with a proper intervals? Does not systick interfere?

adamgreen, 2016-01-26:
>Does not systick interfere?

My suspicion would be that the RTOS sets up the SysTick timer interrupt and that is what actually wakes up the code when the RTOS is linked in.

To someone who knows STM parts better:
* Would the SysTick interrupt as configured by RTX be able to wakeup the device from a stop?
* Is @mfiore02 setting up the RTC as a wakeup source correctly?

mfiore02, 2016-01-27:
I am setting up the RTC wakeup source as specified by the STM32F411 reference guide.

My understanding is that systick fires every 1ms, so it seems pretty clear that systick is not waking up the device. If I configure the RTC wakeup for 30 seconds, my device sleeps for 30 seconds and then wakes up (when RTOS is in the project). If I configure RTC wakeup for 2 minutes, my device sleeps for 2 minutes and then wakes up. Etc, etc.

According to the STM32F411 reference guide,
"The Stop mode is based on the Cortex®-M4 with FPU deepsleep mode combined with peripheral clock gating. The voltage regulator can be configured either in normal or low-power mode. In Stop mode, all clocks in the 1.2 V domain are stopped, the PLLs, the HSI and the HSE RC oscillators are disabled. Internal SRAM and register contents are preserved."

Wouldn't the systick also be getting disabled when the device enters stop mode?

bcostm, 2016-01-27:
Unfortunately we don't have a basic example using deepsleep (stop mode) and wakeup by RTC.

But I took this nice example https://developer.mbed.org/users/Sissors/code/WakeUp/ from @Sissors and I confirm that it works fine on the Nucleo_F401RE and F411RE. I used the online IDE with mbed v112.

There is no need to add the RTOS in the code.

maclobdell, 2016-01-27:
this thread might be of interest: 
https://developer.mbed.org/questions/2201/mbed-sleep-problem-with-RTOS/

at first look, without digging into it, this might be that you are putting the device into a state that it isn't able to be woken up from the wakeup source you have intended, and including the rtos library will configure the systick to fire an interrupt every 1ms which might be waking up the device.

mfiore02, 2016-01-27:
@bcostm thanks for the feedback - I'm taking a look at the library @Sissors wrote now - good to have another datapoint

@maclobdell I'm fairly certain it's not systick as the device doesn't wake up 1ms after it goes to sleep - it sleeps for the configured amount of time

I'm going to spend some time diffing my code and @Sissors code to see what I might be doing wrong

Sissors, 2016-01-27:
While it does not really explain why your code works with RTOS, just to make sure: If you are using a normal NUCLEO board, are you power cycling your MCU using for example the supply current measurement jumper? Core M4s cannot wake from deepsleep when they are being debugged, and those interface ICs like to put them in debug on startup (i know for sure the nxp and freescale ones do, not 100% sure about STM ones. Just resetting is not sufficient).

You can try using an InterruptIn also, that should get it running from stop mode again, which allows you to check if the problem is in the RTC interrupt or in the power-mode code.

mbrudevoldlpd, 2016-02-10:
One issue I note is that you haven't fully initialized "RTC_HandleTypeDef RtcHandle;".  The call into HAL_RTCEx_SetWakeUpTimer_IT needs more than just Instance set.  It needs Lock set to HAL_UNLOCKED.  It may be that you're getting lucky when the RTOS is added and not lucky when it's not.  You can check the return value of HAL_RTCEx_SetWakeUpTimer_IT to make sure it reports success.

0xc0170, 2016-04-13:
@mfiore02 Any update for this issue?

mfiore02, 2016-04-13:
I've switched to using the WakeUp library and, after resolving an issue not related to sleeping, everything is working reliably.

Closing the case.

closed 2016-04-13 by mfiore02