ARMmbed/mbed-os #1514: Proposal to remove target support for GCC_CS  and GCC_CR

sg-, 2016-01-24:
I would like to submit a PR with some house cleaning. From what I've seen most current releases of Eclipse based IDEs are using GCC_ARM. Is anyone specifically using GCC_CS or GCC_CR and a current mbed lib releases? If now I'd like to propose that we depreciate support for these as of Q1 2016. All releases for GCC would then consolidate on the GCC_ARM builds.

0xc0170, 2016-01-25:
Thanks for sharing this before it gets removed. I'll keep an eye on this thread plus the one on mbed.

What we could do for the next release, for exporters, if anyone exports for GCC_CR/CS, we would add a warning or file that its being deprecated and removed in the next xxx revision.

0xc0170, 2016-01-25:
What about lpcxpresso? We got exporters, does it support gcc arm?

toyowata, 2016-01-25:
I believe the LPCXpresso exporter uses GCC_CR (GCC codered) target.

sg-, 2016-01-25:
Is this for redlib or newlib? From the release notes of v7 "Support for ARM’s newlib-nano C library, which can significantly reduce code size, especially for C++ applications" so wondering if the exporters configured for v6 and not v7?

toyowata, 2016-01-26:
By default, the GCC_CR exporter support Newlib, not Newlib-Nano.
Developer can enable Newlib-Nano library in LPCXpresso IDE by following information:
https://www.lpcware.com/content/faq/lpcxpresso/newlib-nano-support

Therefore, existing GCC_CR exporter support all version of LPCXpresso IDE (v6, v7 and v8) and can be switched to use Newlib-Nano library if developer want to.

toyowata, 2016-01-26:
Removing to the GCC_CR means LPCXpresso exporter will no longer support, right?

sg-, 2016-01-26:
No - It means that only v7 and v8 would be supported and use newlib-nano or the stdlib shipped with arms gcc release

toyowata, 2016-01-26:
In my understanding, LPCXpresso exporter (codered.py) uses GCC_CR build target.  How are we going to support LPCXpresso exporter without GCC_CR target then?
https://github.com/mbedmicro/mbed/blob/master/workspace_tools/export/codered.py

sg-, 2016-01-26:
right so that would be updated to use GCC_ARM

0xc0170, 2016-01-28:
I am fine with this proposal. We could carry out this task in the February, send a removal pull request, test it in the online IDE if exporters are functional for lpcxpresso.

sg-, 2016-02-11:
https://github.com/mbedmicro/mbed/pull/1535

Will do much more testing before making the CodeRed/LPCXpresso change to document any deprecation if any exists.

theotherjimmy, 2017-04-26:
Done. GCC_CR and GCC_CS are no longer in the codebase.

closed 2017-04-26 by theotherjimmy