ARMmbed/mbed-os #1525: Set -funsigned-char flag when exporting to GCC

aamcampbell, 2016-02-03:
Here's a minor feature request for exporting to GCC and GCC-based environments (such as KDS and Atmel Studio): add the compiler flag`-funsigned-char` in exported projects by default.
)
The mbed handbook lists the `char` type as being unsigned ([here](https://developer.mbed.org/handbook/C-Data-Types)), but the C standard does not specify, and in GCC `char`s are signed by default (see http://stackoverflow.com/a/2054959/1858923). This leads to errors when exporting projects that assume that `char`s are unsigned, such as [Neil Thiessen's SDFileSystem](https://developer.mbed.org/users/neilt6/code/SDFileSystem/). If we automatically set `-funsigned-char` in exported projects, we reduce the inconsistencies between the mbed online compiler and other target compilers for exported projects.

0xc0170, 2016-02-11:
@aamcampbell Please send PR to workspace_tools/toolchain scripts, with a fix

adamgreen, 2016-02-11:
Pros: Makes GCC builds the same as the online compiler.
Cons: Promotes writing code which isn't really portable.  The correct solution would be for the developer to be explicit with signed/unsigned designation for their char variables (most easily by using the stdint.h types).

0xc0170, 2016-02-11:
Adam, it was stated that mbed library or anything else, hosted on mbed.org and used in the online compiler , have the assumption that char is unsigned. Therefore users wrote the libraries with this assumption, which of course means it's not portable !

I am in favor of consistency, and we should state this in the document (where the info is that char is unsigned in the online compiler).

adamgreen, 2016-02-11:
Go for it.  I just tested with GCC-ARM toolchain and it appears to be unsigned there by default too.

0xc0170, 2016-02-11:
@aamcampbell I would like to know more details where did you have this problem. The one you referenced is for gcc but not specifying the platform.. GCC ARM for instance as Adam pointed out above defaults char to be unsigned. Aren't Atmel and KSD using GCC ARM?

If you can share an example which I can export and get those warnings, would be helpful.

aamcampbell, 2016-02-11:
@0xc0170 Sure. Neil Thiessen's SDFileSystem is the project where I ran into this problem, exported to Kinetis Design Studio, running on a FRDM-K64F. KDS does use arm-none-eabi-gcc, but clearly defaults to signed chars. (See also: https://community.freescale.com/thread/335786)

Steps to reproduce:
- Import this project: https://developer.mbed.org/users/neilt6/code/SDFileSystem_HelloWorld/
- Change lines 5-6 to:

<pre>DigitalIn button(PTC6, PullUp);
SDFileSystem sd(PTE3, PTE1, PTE2, PTE4, "sd", PTE6, SDFileSystem::SWITCH_NEG_NO, 25000000);</pre>

- Export to Kinetis Design Studio, for platform FRDM-K64F
- Set a breakpoint on line 443, `return (resp > 0x00);`
- Compile and run on the target board, with an SD card inserted. Press the user button to start the program. If all goes well, you'll hit the breakpoint with `resp` equal to `0xFF`. Step out of the function, and you'll see that it returned `false` because `0xFF` is interpreted as a negative number.
- Add the compiler flag -funsigned-char to fix the problem.

See also discussion at https://developer.mbed.org/questions/62085/No-response-from-SD-card-on-FRDM-K64F/.

aamcampbell, 2016-02-11:
Also, I'm not quite comfortable submitting a pull request because I'm not familiar with the export scripts and I have no way of testing them. I assume I would add "-funsigned-char" to `common_flags`, but I wouldn't be completely sure.

commit 62b09869 (2016-06-13): Adds -funsigned-char to default gcc args

It would resolve #1525, but for the fact that KDS is used.
commit 685f4ed6 (2016-06-13): Adds -funsigned-char to default gcc args

It would resolve #1525, but for the fact that KDS is used.
closed 2016-06-13 by 0xc0170