ARMmbed/mbed-os #1549: [Export] Missing flags for Seeeduino Arch Pro in Emblocks project.

valeros, 2016-02-15:
Hello!
We have a small problem in platformio/platformio#508.
Looks like linker flags `-u _printf_float` and `-u _scanf_float` are missing in exported project for Emblocks, so `printf` doesn't work correctly for float values. Is this a bug?
Thanks!

0xc0170, 2016-02-15:
This seems like a problem in the toolchain file, where for gcc: https://github.com/mbedmicro/mbed/blob/20948ad920372de6b5ff5bf2d4f2237f3730b470/workspace_tools/toolchains/gcc.py#L192 . The list seems to be not consistent, we shall have a look

cc @sg-  @EmBlocks

sg-, 2017-01-16:
Toolchain flags have been unified and are shared between a mbed tools and exported projects.

closed 2017-01-16 by sg-