ARMmbed/mbed-os #1561: [TOOLCHAIN_GCC_ARM] Using __get_MSP() in _sbrk()

betzw, 2016-02-23:
@0xc0170 @bcostm 

Yesterday I was giving a closer look to file ```libraries/mbed/common/retarget.cpp``` and especiallaly to the lines from line [474](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/common/retarget.cpp#L474) to line [492](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/common/retarget.cpp#L492) where function ```__sbrk()``` gets defined for ```TOOLCHAIN_GCC_ARM```, which looks like this:
```
// Dynamic memory allocation related syscall.
extern "C" caddr_t _sbrk(int incr) {
    static unsigned char* heap = (unsigned char*)&__end__;
    unsigned char*        prev_heap = heap;
    unsigned char*        new_heap = heap + incr;
https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/common/retarget.cpp#L482
#if defined(TARGET_ARM7)
    if (new_heap >= stack_ptr) {
#elif defined(TARGET_CORTEX_A)
    if (new_heap >= (unsigned char*)&__HeapLimit) {     /* __HeapLimit is end of heap section */
#else
    if (new_heap >= (unsigned char*)__get_MSP()) {
#endif
        errno = ENOMEM;
        return (caddr_t)-1;
    }

    heap = new_heap;
    return (caddr_t) prev_heap;
}
```
My point here refers to line [484](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/common/retarget.cpp#L484) where the availability of further heap space is checked by comparing the ```new_heap``` end with the position of the current stack pointer (```__get_MSP()```).
In my eyes, this is not a very secure/good way to check the availability of further heap space as in case at the moment of the call we are in the situation of low stack usage the check might pass and the memory added to the heap, while further on the stack might grow beyond its current value (i.e. as we are growing down ```__get_MSP()``` might return a minor value) and such invade the already allocated heap space ...
Furthermore, even if less important, the usage of ```__get_MSP()``` as heap limit implies a certain memory layout in which the heap and stack must be placed in the same memory bank and where the heap must be located immediately before the stack region, which on some platforms (e.g. STM32L476RG) limits the possible choices.

A solution to this issue might be to implement ```__sbrk()``` as it is done for ```TARGET_CORTEX_A``` (see line [482](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/common/retarget.cpp#L482)) also in case of a Cortex-M, i.e. using ```__HeapLimit``` to delimit the heap space. But this requires a review of all linker scripts for the GNU toolchain for most likely all platforms.

What do you think?

bcostm, 2016-02-23:
Thanks for this analysis. I don't know why today we can't use the __HeapLimit as done on the Cortex_A? I am not an expert of this. Maybe ARM can help ? If a fix is found for one or two platforms, for example the F401 and L476, we can help afterwards to deploy this change to all other ST platforms.

0xc0170, 2016-02-23:
cc @adamgreen

adamgreen, 2016-02-23:
The current version is probably done for maximum flexibility since it doesn't require the user to figure out beforehand how much memory they want to dedicate to the stack and heap. The obvious cost of this flexibility is that the stack can grow downward and collide with the heap and it requires the heap and stack grow towards each other in the same area of memory.  

Having a HeapLimit forces the user to figure out beforehand how big they want their stack and heap to be.  What does the user have to modify to grow the heap/stack limits in your plan?  The linker script? 

For most platforms just adding a HeapLimit doesn't really solve anything anyway since it only limits heap growth and does nothing to stop the stack from growing down into the heap.  To really make it work, you would want to pre-allocate space for the stack as well and place it somewhere in memory where it could never grow down into the heap or anything else.

The best solution might be to pull sbrk() out of retarget.cpp and place it in its own sbrk.o object file so that it can be replaced completely by any platform that wants to use a different memory model.  Could also have a DEVCE_* macro in device.h which can disable the inclusion of the one in retarget.cpp for a particular platform.

@0xc0170 What does the online compiler do?

betzw, 2016-02-23:
Well, the HeapLimit would also lower the probability that the heap gets corrupted by the stack as its growth will be limited. Maybe not by chance in mbedOS the memory layout has been changed to have the stack being the very first region (assuming that the vector tables will also be moved further up in future)!?!

adamgreen, 2016-02-23:
Yes, if the HeapLimit is quite small but if you make that the default then people will start complaining about their programs now having allocation failures when currently they don't.  I don't like setting HeapLimits because I don't know what they should be.  Each project will have a different split between stack and heap usage.

I do agree that allowing some flexibility in placing the heap in a different (or even across several) bank(s) might be nice on some platforms.

betzw, 2016-02-23:
No doubt about that limiting the size of the heap might create problems for some applications, as for other applications the current approach might not work out at all (e.g. when allocating a lot of memory initially, which then gets partially freed, before entering computations which use recursive functions) ...
Not sure, if there is a "right" solution for mbed-classic. I believe it is a trade-off which has to take into account all the design goals of mbed-classic and their respective priorities.
As far as I understand, e.g. in mbed-os you need to specify only a stack limit and the heap will catch all the rest of the memory which has been left by the linker, which might be a good compromise. 
(Also in mbed-os there is no support for multiple non-consecutive memory banks.)

adamgreen, 2016-02-23:
If you set HeapLimit too low, the user will get alloc failures when there is still memory available.  If you set it too high then they are in the same situation as today where a recursive function could collide with the heap.  The correct value depends on how much stack the developer needs.

Putting a defined stack at the bottom of valid RAM is one way to catch such stack overflows.  Another is to use the MPU (on CPUs which have one) to place a guard block of 32-bytes between the stack and other areas of memory so that a stack overflow generates a memory fault.  But that is something that the developer should do themselves as they might have other uses for the MPU where they would need to add this into the mix.

betzw, 2016-02-23:
Btw, as far as I know the online compiler uses limits at least with respect to the heap.
@0xc0170 might confirm.

adamgreen, 2016-02-23:
At least for the LPC1768 target that doesn't appear to be the case:
https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/TOOLCHAIN_ARM_STD/sys.cpp
https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/TOOLCHAIN_ARM_MICRO/sys.cpp

betzw, 2016-02-24:
Well, as far as I know at least in TOOLCHAIN_ARM_MICRO ```__user_setup_stackheap``` gets never called and therefore is removed from the binary by the linker.
Anyway, I have the impression that there is no real **general** guideline regarding the memory layout in mbed-classic. Pls. correct me if I am wrong!

sg-, 2016-02-24:
The right solution for the mbed SDK has been letting the stack and heap collide. https://developer.mbed.org/handbook/Memory-Model

0xc0170, 2016-02-24:
> The right solution for the mbed SDK has been letting the stack and heap collide. https://developer.mbed.org/handbook/Memory-Model

:+1:

betzw, 2016-02-24:
@siorpaes

betzw, 2016-02-25:
So my question now becomes: considering that the future of mbed should be mbed-os, is it really worthwhile to align all mbed-classic platforms a/o tool-sets which do currently not implement this memory model to become compliant with it?

sg-, 2016-02-26:
Yes - All mbed Classic platforms should implement this memory model. https://developer.mbed.org/handbook/Memory-Model We need to create a test for this. Anyone interested??

JojoS62, 2016-02-26:
Could you please add in the documentation that the RAM section must have space for the copy of the interrupt vector table?

0xc0170, 2016-02-28:
@JojoS62 Please some details, where should it be added? The picture there in the link @sg- referenced above?

0xc0170, 2016-02-28:
> We need to create a test for this. Anyone interested??

@sg- Can you create an issue with some info about the tests, someone (even me) can find time to implement them

sg-, 2016-02-28:
@JojoS62 great recommendation. I'll update the ascii art and handbook page text about this.

JojoS62, 2016-02-28:
Thanks for regarding this. 
Of course it's only an issue when you use mbed in an offline toolchain with generated standard linker scripts, but then it's hard to find the reason why interrupts don't work. 
Thanks to Erik Olieman, he explained me this. 

> Am 28.02.2016 um 23:53 schrieb Sam Grove <notifications@github.com>:
> @JojoS62 great recommendation. I'll update the ascii art and handbook page text about this.
>

betzw, 2016-02-29:
@sg- can you pls. provide me with a reference implementation of this memory model for TOOLCHAIN_ARM_MICRO.

betzw, 2016-03-18:
@sg- here might be a better place for the discussions you have kicked-off in #1554 and #1536 
also because I am still waiting for some answers (see above) ...

pan-, 2016-06-28:
@sg- @mjs-arm The [Memory model handbook](https://developer.mbed.org/handbook/Memory-Model) is confusing to me.

```
Obviously, looking at this, it is possible to for your heap and stack to collide, which is never 
going to end well! If that happens, you have hit the limit of your RAM, and need to find out 
ways to help ensure your program uses less memory.

To try and help you prevent detect when this would happen, the routines that allocate memory 
on the heap (new, malloc and friends) tell you when you don't have enough memory. Instead 
of passing you a pointer to the newly allocated block, they pass you the value NULL.
```

I wonder how malloc and friends can prevent any collision right now. 

Since `malloc` is called in thread mode and the main stack is only used by interrupts (thanks to the RTOS) then it is likely that the main stack is empty and MSP very close to the top of the RAM when `_sbrk()` allocate memory.

This can go really really bad on platforms requiring a lot of memory for their main stack. For instance, on NRF51 the softdevice can eat 1536 bytes of main stack. If `__sbrk` use the MSP as a reference for the end of the stack, then it let a lot of room for bad allocations and heap corruption.

@adamgreen There is no need to set explicitly a heap limit, the stack size is enough because the heap limit is usually equal to `ORIGIN(RAM) + LENGTH(RAM) - Stack_Size`. 

It is also possible to keep the current behavior while allowing targets with a heap limit defined to take advantage of it:

```c++
extern "C" MBED_WEAK uint32_t  __HeapLimit = 0;

unsigned char* get_heap_limit() {
    if (__HeapLimit) {
        return (unsigned char*) &__HeapLimit;
    }

#if defined(TARGET_ARM7)
    return stack_ptr;
#elif !defined(TARGET_CORTEX_A)
    return (unsigned char*)__get_MSP();
#endif
}

extern "C" caddr_t _sbrk(int incr) {
    static unsigned char* heap = (unsigned char*)&__end__;
    unsigned char*        prev_heap = heap;
    unsigned char*        new_heap = heap + incr;

    if (new_heap >= get_heap_limit()) {
        errno = ENOMEM;
        return (caddr_t)-1;
    }

    heap = new_heap;
    return (caddr_t) prev_heap;
}
```

Avnet-Silica, 2017-10-10:
I understood all the discussion and prons and cons, but four more things:
- in your solution to check heap top, if you allocate something in the heap, and the stack will grow later to use memory values below the actual heap top, you will overwrite the data allocated in the heap, and you have no way to know this had happened except for random behavior of your device;
- having a known error ( heap memory finished ) to manage is much better than having random failures of the device
- if you have a known heap size, the linker can check if there is enough memory at link time; when you make a change on the code, for example adding a big structure, you get at link time an error if you exceed the memory available, not at run time
- as a programmer which wants to prevent random failures, I want to do the largest number of checks at compile time; comparing heap top and actual stack pointer is too dependent of the actual run conditions, making complete test almost impossible
Also, having a heap size will make better use of "fragmented" memory like the one of ST processors, where you can put the heap on a separate piece of memory

mjs-arm, 2017-10-11:
Picking up on two of the (much) earlier comments:

*) Placing a 32 byte guard block below each stack (assuming mpu hw) is not sufficient it itself.

To prevent accidental collisions due to insufficient RAM leading to heap and stack collision the guard block needs to be large enough that a stack frame cannot straddle  the guard.

But from a security perspective (i.e. preventing a variety of attacks against a device deliberately exploiting stack overrun) the problem is more involved. For background on stack clash attack, for example, refer to https://access.redhat.com/security/vulnerabilities/stackguard  . This article is written from a linux perspective but the attack scenario holds equally for other OSs.  Mitigation for this kind of attack involves both guard blocks and compiler support (refer to https://gcc.gnu.org/ml/gcc-patches/2017-07/msg00556.html for an outline of the compiler approach). Note how the compiler mitigation is coupled to the guard block size used by the OS.

*) Placing the stack at the bottom of memory. Is a strategy that only works if there is one stack. Current mbed-os is threaded and hence has multiple stacks... all of which can overflow.

40Grit, 2017-10-11:
> But from a security perspective (i.e. preventing a variety of attacks against a device deliberately exploiting stack overrun) 

Still can't understand why a core would ever be designed such that return addresses are pushed onto the same stack that data is pushed to.

Additional informational reference: 
https://embeddedgurus.com/state-space/2014/02/are-we-shooting-ourselves-in-the-foot-with-stack-overflow/

betzw, 2018-01-12:
#deadbeef

closed 2018-01-12 by betzw