ARMmbed/mbed-os #1562: fopen(filepath, "rb") doesn't seem to work

justbuchanan, 2016-02-24:
Whenever I try to open a file in binary mode, `fopen()` returns `nullptr`, but `errno` is set to `0`.  I have verified that the file I'm opening exists and it opens just fine in text mode when I pass `"r"` instead of `"rb"`.

adamgreen, 2016-02-25:
Don't use the 'b' (binary) option with mbed.  Some of the mbed supported Standard C Libraries just strip it off and others pass it down to lower level file system drivers which then fail on it.  I have never seen any evidence that a mbed supported C Library will actually treat 'text' files any different than 'binary' ones just like most Posix systems.

justbuchanan, 2016-02-27:
Ok, makes sense.  I'll just stick with "r" for the mode.  Thanks!

closed 2016-02-27 by justbuchanan

adamgreen, 2016-02-27:
@justbuchanan Which toolchain and FileSystem did you hit this with?  I have seen it before with GCC/newlib and LocalFileSystem.  Is that what you were attempting to use as well?

justbuchanan, 2016-03-02:
@adamgreen, yeah this was with GCC and LocalFileSystem on an MBED LPC1768.

adamgreen, 2016-03-02:
Thanks for the info.  I think I have a fix that will stop that from happening.  I am planning to do some file system related PRs over the next couple of weeks.  I will add this to my list.