ARMmbed/mbed-os #1574: IAR error in command line not reported while compiling

adustm, 2016-02-29:
Hello,
The following command line is creating an error
build.py -v -x -c -t IAR -m DISCO_F746NG,NUCLEO_F746ZG -j 4

``[DEBUG]` Command: iasmarm --cpu Cortex-M7F -DTARGET_STM32F746NG -DTARGET_STM32F746 -DARM_MATH_CM7 -DTARGET_CORTEX_M -D__CORTEX_M7 -D__FPU_PRESENT=1 -DTARGET_DISCO_F746NG -DTARGET_RTOS_M4_M7 -DMBED_BUILD_TIMESTAMP=1456501269.49 -DTARGET_STM -DTARGET_M7 -DTARGET_STM32F7 -DTOOLCHAIN_IAR -D__MBED__=1 -I.\mbed\targets\cmsis -I.\mbed\libraries\mbed\targets\cmsis\TARGET_STM -I.\mbed\libraries\mbed\targets\cmsis\TARGET_STM\TARGET_STM32F7 -I.\mbed\libraries\mbed\targets\cmsis\
TARGET_STM\TARGET_STM32F7\TARGET_DISCO_F746NG -I.\mbed\libraries\mbed\targets\cmsis\TARGET_STM\TARGET_STM32F7\TARGET_DISCO_F746NG\TOOLCHAIN_IAR -I.\mbed\libraries\mbed\targets\cmsis\TOOLCHAIN_IAR -o .\mbed\build\mbed\.temp\TARGET_DISCO_F746NG\TOOLCHAIN_IAR\TARGET_STM\TARGET_STM32F7\TARGET_DISCO_F746NG\TOOLCHAIN_IAR\startup_stm32f746xx.o .\mbed\libraries\mbed\targets\cmsis\TARGET_STM\TARGET_STM32F7\TARGET_DISCO_F746NG\TOOLCHAIN_IAR\startup_stm32f746xx.S
[DEBUG] Return: 3
Traceback (most recent call last):
  File "..\mbed\workspace_tools\build.py", line 234, in <module>
    macros=options.macros)
  File "C:\MCDGitW601\mbed\workspace_tools\build_api.py", line 445, in build_mbed_libs
    raise e
ToolException
``
I had to type the command line manually to discover that it was an option of the --cpu that was not supported:
``Error in command line: Invalid argument to option '--cpu'
``
Could you make sure this error is reported ?

Kind regards

0xc0170, 2016-02-29:
cc @bridadan

bridadan, 2016-02-29:
@adustm thanks for finding this issue! Just to be clear, you added IAR to the supported toolchains of both the DISCO_F746NG and the NUCLEO_F746ZG before running build.py, correct? I ran from the latest master and I received a "toolchain not supported" message by both of those targets.

I went ahead and added those toolchains to the targets.py and I was able to replicate you issue. The error isn't being propagated because the error isn't being printed to stderr by IAR, its being printed to stdout. You can see on this line that we aren't looking for errors on stdout currently: https://github.com/mbedmicro/mbed/blob/master/workspace_tools/toolchains/__init__.py#L40. `run_cmd` returns `stdout`, `stderr`, and the return code (in that order).

@0xc0170 Should we add support for print stdout as well? Or does this sound like something we should bring up with IAR?

adustm, 2016-03-03:
> Just to be clear, you added IAR to the supported toolchains of both the DISCO_F746NG and the NUCLEO_F746ZG before running build.py, correct? I ran from the latest master and I received a "toolchain not supported" message by both of those targets.
Yes correct. @0xc0170 has removed IAR for those boards as there was an error.

bridadan, 2016-03-03:
@adustm Ok, thanks for clarifying.

@0xc0170 It seems to me like IAR should be printing this on stderr instead of stdout. But seeing how we can't make that change ourselves, should we add support for printing stdout? Maybe if the verbose flag is set?

0xc0170, 2016-03-26:
> It seems to me like IAR should be printing this on stderr instead of stdout. But seeing how we can't make that change ourselves, should we add support for printing stdout? Maybe if the verbose flag is set?

Yes, even for IAR to get stdout for the error. We can reach out to IAR why stdout is used.

0xc0170, 2016-04-11:
This was fixed via https://github.com/mbedmicro/mbed/pull/1604

closed 2016-04-11 by 0xc0170