ARMmbed/mbed-os #1583: Summary table should be exportable to text file

PrzemekWirkus, 2016-03-02:
Currently we can export reports to HTML and JUnit using ```--report-XXX``` switch.
We should be also able to export to pure text file like this:
```
$ singletest.py .... --report-text summary_table.txt
```

0xc0170, 2016-03-02:
+1. I would like to encourage sharing test results for pull requests, this would be nice to just paste the text from txt file.

bcostm, 2016-03-02:
I have requested this change in order to put more easily the test results in pull requests comments. 
The idea is to save the output which is today displayed in the DOS window inside a text file. This way it will be easier to retrieve these information and copy/paste them in pull requests for example. Because sometimes I close by inadvertance the DOS windows and the text result is lost.

bridadan, 2016-03-02:
This should be quite doable. A while ago I refactored the test reporting a bit. It's still not perfect, but it should be easier to add these new reporters.

The key bits are here: https://github.com/mbedmicro/mbed/blob/master/workspace_tools/test_exporters.py

You need to add a `ResultExporterType` at the top of the script. You can see how to implement a new exporter by looking at the `JUNIT` (not `JUNIT_OPER`) report type.

You can then see how the ResultExporter is used but singletest.py here: https://github.com/mbedmicro/mbed/blob/master/workspace_tools/test_api.py#L1511

That should be enough info to get someone started. I don't think I'll be able to write this at the moment, but if no one picks this up I might revisit this.

0xc0170, 2016-03-03:
> That should be enough info to get someone started. I don't think I'll be able to write this at the moment, but if no one picks this up I might revisit this.

I think it's up to us Brian :-)

bridadan, 2016-03-03:
Most likely, just thought I'd leave that info up there in case anyone got motivated and wanted to take a crack at it :)

PR #1873 (2016-07-06): Adds a simple text file report format
Should resolve #1583
commit 6906b97b (2016-07-06): Adds a simple text file report format

Should resolve #1583
closed 2016-07-06 by sg-