ARMmbed/mbed-os #1599: Online compiler error after c6d2c81

neilt6, 2016-03-08:
The [mbed-dev library](https://developer.mbed.org/users/mbed_official/code/mbed-dev/) is failing to compile using the [Online Compiler](https://developer.mbed.org/compiler/) following commit c6d2c81:

`Error: Identifier "O_BINARY" is undefined in "mbed-dev/common/retarget.cpp", Line: 125, Col: 22`

0xc0170, 2016-03-08:
@neilt6 Thanks for reporting, will have a look at this

cc @adamgreen

adamgreen, 2016-03-08:
@neilt6 Thanks for catching my oops.  I will submit fix soon.

0xc0170, 2016-03-08:
@neilt6 Please retest, and close if resolved

0xc0170, 2016-03-08:
Thanks @adamgreen for the instant fix

neilt6, 2016-03-08:
Confirmed, [mbed-dev](https://developer.mbed.org/users/mbed_official/code/mbed-dev/) compiles and runs fine now. Thanks @adamgreen!

closed 2016-03-08 by neilt6