ARMmbed/mbed-os #1617: Exported project - should contain readme with some useful information

0xc0170, 2016-03-17:
To be able to reproduce the issue locally, we should provide more information about the tools which are used for exporting.

For instance, if progen is active for an exporter, I would like to see the progen version, so if the build fails and in the online IDE it does not, report it here with the specific progen version.

I could fetch that version, write a test for this failure, fix it and deploy a new fixed version.

We shall have a look further for what kind of information might be useful for reporting problems with exporters.

```
// readme.txt
progen = True // False if not active
progen_version = 0.8.10 //if progen is not active this is not here or invalid
```

0xc0170, 2016-03-17:
@sg- @bridadan

bridadan, 2016-03-18:
Makes sense to me!

PR #1636 (2016-04-05): Export - add readme with exporters information
Adding readme with exporters information. What should be the format ? Should it be json, yaml or just simple text file with predefined format.

Current format in exporter.txt:
```
project_generator = true/false # if progen is used
# if progen is true, then version is added
project_generator version = x.x.x
```

Should fix #1617. For more info about getting the module version: http://python-packaging-user-guide.readthedocs.org/en/latest/single_source_version/

@bridadan @sg-
closed 2016-04-05 by 0xc0170