ARMmbed/mbed-os #1618: Cannot set the RTC on the NUCLEO-F746ZG

loopsva, 2016-03-18:
I can properly set the RTC on the DISCO-F746NG, but not on the NUCLEO-F746ZG

...kevin

0xc0170, 2016-03-18:
@bcostm @adustm

bcostm, 2016-03-18:
I checked the Nucleo_display_time example on both DISCO_F746 and NUCLEO_F746 and it works. I used mbed lib **v116**. Which code do you use ? Can you share it ?

loopsva, 2016-03-18:
I cannot share this code publicly.  I will try the Nucleo_display_time example and will get back to you on Tuesday.  
Thanks,...kevin

loopsva, 2016-03-22:
My issue had to do with the mktime structure.  I was not subtracting 1900 from the year, creating a -1 return value for total seconds.  Sorry for the false alarm.

closed 2016-03-22 by loopsva

0xc0170, 2016-03-23:
@loopsva For the future, please add code snippet  to reproduce the problem.