ARMmbed/mbed-os #1619: mbed v116 breaks F746Nx serial RX interrupts

loopsva, 2016-03-18:
Hello,

When migrating the mbed library from v115 to v116, USB serial RX interrupts stopped working (kind of).

1. Receiving single characters is ok, the RX serial interrupt works fine.

2. When receiving character bursts, the issue appears. Example is a PC's UP arrow. It sends a 3 character sequence of:

(ESC) [ A 

When the UP arrow burst occurs, the CPU freezes. All other RTOS tasks also freeze.

...kevin

0xc0170, 2016-03-18:
@bcostm @adustm

bcostm, 2016-03-18:
The official mbed USB library is not supported on the F7 platforms... Which USB library are you using ?

loopsva, 2016-03-18:
Sorry,I didn't mean the official USB library.  I meant the CMSIS-DAP USB interface to the local CPU.  I believe that is USART3 for the F746.
More info:
Baud rate is 230400.  I do have overhead in processing each character inside the ISR, including checking for special chars (CR, LF, BS, etc.), adding char to line buffer, echoing char out the UART TX.  Maybe this is causing a RX overrun (no FIFO?), but I've been using this code for years using various mbed CPUs.  All was ok until v116 of mbed.  The K64F is still ok with v116, but it has a receive FIFO.

loopsva, 2016-03-22:
More testing,
I have determined that it is not a baudrate-receive overflow problem.  I slowed the UART to 9600 and the issue is still there.  What I did find is that an interaction between receiving the character burst (ESC) [ A, and echoing the characters back out of the console.  When sending out the echo character, I also send out a couple escape sequences with it (basically making the echoed output character green).  If I just echo back the single character, the mbed v116 works fine.  However, adding the escape sequences breaks v116 on the F746 (is ok on a K64F). v115 is ok with this on a F746.

#define DOGREEN     "\033[0;32;2m"
#define DONONE      "\033[0m"

//inside the ISR
//...code
if(LocalEcho) pc.printf("%s%c%s", DOGREEN, pcRxBuffer[pcRxQty - 1], DONONE);
//...code

0xc0170, 2016-03-23:
@loopsva Can you create a short code snippet to reproduce this problem locally, that would help. One more thing to look at, make a diff of changes between v115 and v116 in the F7 HAL.

adustm, 2016-11-22:
Hello, is this issue still alive ?

sg-, 2017-01-16:
@loopsva please create an example program for this issue if it still exists and reopen.

closed 2017-01-16 by sg-