ARMmbed/mbed-os #1632: Support EthernetInterface for STM32F746

sandeepach, 2016-03-26:
Support EthernetInterface for STM32F746G. I get too many compile errors and resolving them is not easy.

0xc0170, 2016-03-26:
Is this a feature request? You can share your branch, might help someone to start.

cc @bcostm @adustm

sandeepach, 2016-03-28:
Hello Martin
I dont have a working branch yet. The EthernetInterface standard library from mbed official does not support STM32F7 yet. Hard to get the Ethernet communication working with a mbed only setup.

bcostm, 2016-09-09:
Ethernet on F7 devices has been done in PR #2473 

Can you please close this issue ?

bogdanm, 2016-09-09:
@sandeepach: can you please check if the issue is fixed?

sandeepach, 2016-09-17:
Hello
I have no way to check this directly. 
Only the github is having the ethernet feature. Is there a mbed library i can download which has net/ipv4 feature. 
Regards
Sandeep

bcostm, 2016-11-14:
Hi,
We have checked this example [https://developer.mbed.org/teams/ST/code/mbed-os-tcp-server-example/](https://developer.mbed.org/teams/ST/code/mbed-os-tcp-server-example/) and the Ethernet works on this board. Can you please check it also ?

adustm, 2017-01-03:
Hello, 
As Ethernet is supported for F7 familly, are we happy to close this issue ?
Cheers
Armelle

bridadan, 2017-01-03:
Thanks for following up @adustm, I'll go ahead and close this.

closed 2017-01-03 by bridadan