ARMmbed/mbed-os #1634: STM32 RTC backup registers get reset when setting the clock

campag, 2016-03-26:
I started to use the backup registers on Nucleo F401 with the HAL library, but notice that if I set the time with mbed, my data is also erased - mbed's set_time() in rtc_time.c calls rtc_init() before rtc_write().

However STM32F4's (& I suppose others) rtc_int() in rtc_api.c erases the backup registers. If I comment out
```
/*
        __HAL_RCC_BACKUPRESET_FORCE();
        __HAL_RCC_BACKUPRESET_RELEASE();
*/
```
then my backup register data is untouched.

Is there a reason those lines are there?  After all, mbed api doesn't use the backup registers, and they appear to be reset anyway on power-up without Vbat.

0xc0170, 2016-03-26:
cc @bcostm @adustm

bcostm, 2016-03-31:
We made some trials and these 2 lines are only needed when the LSI is used. For the LSE, they can be removed. What we are going to do is to move these 2 lines in the LSI section only.

campag, 2016-03-31:
Thanks for looking at this. Current Nucleo production seems to be fitted with LSE, so the fix is fine there.

I guess it won't be of use to those with older models & LSI though, and somewhat ironically those are the ones who will be setting the clock most often.

For those wanting to play with the backup registers under mbed, I put code on my notebook: [https://developer.mbed.org/users/gregeric/notebook/using-stm32-rtc-backup-registers/](url)

0xc0170, 2016-04-13:
@campag is this resolved, or still relevant?

campag, 2016-04-13:
Resolved thanks. Will close but one quick question - where should I report small issue with the STM HAL library (re ADC from temp & Vbat). Doesn't seem appropriate here?

closed 2016-04-15 by campag

bcostm, 2016-04-22:
For the support of ADC internal channels (Vbat, Vref, Temp) you can enter an issue. We have started to doing it on mbedOS. It could be easily ported to mbedClassic.