ARMmbed/mbed-os #1644: nRF51-DK export to uVision is broken

janjongboom, 2016-04-01:
Exporting simple program (like UriBeacon) to uVision does not create a buildable uvproj file (I have uVision 5, but I don't think that matters...), gives a few hundred errors.

0xc0170, 2016-04-01:
Can you test simple demo exported form this repository? Seems like my uvision does not want to fetch new devices at the moment :(

```
 python workspace_tools\build.py -m NRF51822 -t ARM -j 4
python workspace_tools\project.py -m NRF51822 -i uvision -p 47-b
```

What type of errors are you seeing? Is it missing includes or some settings are wrong?

sg-, 2016-04-01:
Also make sure the program in your workspace is updated to the latest versions of the mbed library.

janjongboom, 2016-04-04:
This program (https://developer.mbed.org/users/janjongboom/code/BLE_URIBeacon2/), all up to date dependencies, exported to uVision. 

Here's an excerpt of the build log:

```
          BLE_GAP_CONN_SEC_MODE_SET_OPEN(&cccd_md.read_perm);
nRF51822/source/btle/custom/custom_helper.cpp(228): error:  #20: identifier "ble_gatts_char_md_t" is undefined
      ble_gatts_char_md_t char_md = {0};
nRF51822/source/btle/custom/custom_helper.cpp(240): error:  #20: identifier "ble_gatts_attr_md_t" is undefined
      ble_gatts_attr_md_t attr_md = {0};
nRF51822/source/btle/custom/custom_helper.cpp(245): error:  #20: identifier "BLE_GATTS_VLOC_STACK" is undefined
      attr_md.vloc = BLE_GATTS_VLOC_STACK;
nRF51822/source/btle/custom/custom_helper.cpp(252): error:  #20: identifier "BLE_GAP_CONN_SEC_MODE_SET_OPEN" is undefined
                  BLE_GAP_CONN_SEC_MODE_SET_OPEN(&attr_md.read_perm);
nRF51822/source/btle/custom/custom_helper.cpp(255): error:  #20: identifier "BLE_GAP_CONN_SEC_MODE_SET_ENC_NO_MITM" is undefined
                  BLE_GAP_CONN_SEC_MODE_SET_ENC_NO_MITM(&attr_md.read_perm);
nRF51822/source/btle/custom/custom_helper.cpp(258): error:  #20: identifier "BLE_GAP_CONN_SEC_MODE_SET_ENC_WITH_MITM" is undefined
                  BLE_GAP_CONN_SEC_MODE_SET_ENC_WITH_MITM(&attr_md.read_perm);
nRF51822/source/btle/custom/custom_helper.cpp(261): error:  #20: identifier "BLE_GAP_CONN_SEC_MODE_SET_SIGNED_NO_MITM" is undefined
                  BLE_GAP_CONN_SEC_MODE_SET_SIGNED_NO_MITM(&attr_md.read_perm);
nRF51822/source/btle/custom/custom_helper.cpp(264): error:  #20: identifier "BLE_GAP_CONN_SEC_MODE_SET_SIGNED_WITH_MITM" is undefined
                  BLE_GAP_CONN_SEC_MODE_SET_SIGNED_WITH_MITM(&attr_md.read_perm);
nRF51822/source/btle/custom/custom_helper.cpp(274): error:  #20: identifier "BLE_GAP_CONN_SEC_MODE_SET_OPEN" is undefined
                  BLE_GAP_CONN_SEC_MODE_SET_OPEN(&attr_md.write_perm);
nRF51822/source/btle/custom/custom_helper.cpp(277): error:  #20: identifier "BLE_GAP_CONN_SEC_MODE_SET_ENC_NO_MITM" is undefined
                  BLE_GAP_CONN_SEC_MODE_SET_ENC_NO_MITM(&attr_md.write_perm);
nRF51822/source/btle/custom/custom_helper.cpp: 30 warnings, 30 errors
compiling ble_conn_params.cpp...
BLE_API/ble/blecommon.h(81): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __BLE_COMMON_H__
BLE_API\ble/BLEProtocol.h(69): warning:  #1-D: last line of file ends without a newline
  #endif /* __BLE_PROTOCOL_H__ */
BLE_API/ble/GapAdvertisingData.h(498): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GAP_ADVERTISING_DATA_H__
BLE_API/ble/GapAdvertisingParams.h(120): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GAP_ADVERTISING_PARAMS_H__
BLE_API/ble/GapScanningParams.h(68): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GAP_SCANNING_PARAMS_H__
BLE_API/ble/GapEvents.h(46): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GAP_EVENTS_H__
BLE_API/ble/SafeBool.h(114): warning:  #1-D: last line of file ends without a newline
  #endif /* BLE_API_SAFE_BOOL_H_ */
BLE_API/ble/FunctionPointerWithContext.h(212): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef MBED_FUNCTIONPOINTER_WITH_CONTEXT_H
BLE_API/ble/CallChainOfFunctionPointersWithContext.h(238): warning:  #1-D: last line of file ends without a newline
  #endif
BLE_API/ble/deprecate.h(26): warning:  #1-D: last line of file ends without a newline
  #endif
BLE_API/ble/Gap.h(1400): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GAP_H__
BLE_API/ble/UUID.h(227): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __UUID_H__
BLE_API/ble/SecurityManager.h(331): warning:  #1-D: last line of file ends without a newline
  #endif /*__SECURITY_MANAGER_H__*/
BLE_API/ble/GattAttribute.h(81): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GATT_ATTRIBUTE_H__
BLE_API/ble/GattCallbackParamTypes.h(93): warning:  #1-D: last line of file ends without a newline
  #endif /*__GATT_CALLBACK_PARAM_TYPES_H__*/
BLE_API/ble/GattCharacteristic.h(547): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GATT_CHARACTERISTIC_H__
BLE_API/ble/GattService.h(86): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GATT_SERVICE_H__
BLE_API/ble/GattServerEvents.h(39): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GATT_SERVER_EVENTS_H__
BLE_API/ble/GattServer.h(484): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __GATT_SERVER_H__
BLE_API/ble/ServiceDiscovery.h(164): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __SERVICE_DISOVERY_H__
BLE_API/ble/CharacteristicDescriptorDiscovery.h(99): warning:  #1-D: last line of file ends without a newline
  #endif // ifndef __CHARACTERISTIC_DESCRIPTOR_DISCOVERY_H__
[... 34 lines cut ...]
nRF51822/source/nordic_sdk/components/ble/common/ble_conn_params.cpp(352): warning:  #12-D: parsing restarts here after previous syntax error
              break;
nRF51822/source/nordic_sdk/components/ble/common/ble_conn_params.cpp(353): error:  #169: expected a declaration
      }
nRF51822/source/nordic_sdk/components/ble/common/ble_conn_params.cpp(379): warning:  #12-D: parsing restarts here after previous syntax error
[... cut ...]
nRF51822/source/nordic_sdk/components/ble/common/ble_conn_params.cpp(386): error:  #169: expected a declaration
  }
nRF51822/source/nordic_sdk/components/ble/common/ble_conn_params.cpp(188): warning:  #177-D: function "conn_params_negotiation" was declared but never referenced
  static void conn_params_negotiation(void)
nRF51822/source/nordic_sdk/components/ble/common/ble_conn_params.cpp: 27 warnings, 21 errors
".\build\BLE_URIBeacon2.axf" - 688 Error(s), 583 Warning(s).
Target not created.
Build Time Elapsed:  00:00:15
```

janjongboom, 2016-04-05:
For K64F it worked. Let me try nrf51.

janjongboom, 2016-04-05:
nrf51 without BLE lib works as well.

0xc0170, 2016-04-05:
If you open the .uvproj file in your editor , the one which fail, can you see there:

```
          <RegisterFilePath>Freescale\Kinetis\</RegisterFilePath>
          <DBRegisterFilePath>Freescale\Kinetis\</DBRegisterFilePath>
```

This is a leftover I have to clean. If the project contains those 2 lines, it is generated by progen. Otherwise it is the old exporter.

janjongboom, 2016-04-07:
Another issue reported on the forum: https://developer.mbed.org/questions/68316/mbed-project-on-Keil-uVision5-has-compil/

0xc0170, 2016-04-11:
@janjongboom There are 2 PR to BLE repositories (nordic-sdk and ble-nrf51822), which fixes this. The problem is not in the exporters, but in the files. There's a conflict of headers (ble.h), which was reported earlier. This causes the build system within uVision (and other IDE) to fail. 

The fixes should be merged soon, and integrated. Btw, we will update to the latest exporters (progen update as well).

We should keep this open until ble repositories are up to date, and resolves this issue.

cc @pan-

pan-, 2016-04-12:
I guess it ~~is fixed~~ works now. 
The issue lies in our build system which include every folder in a project. 
The entry point of ble library is `ble/BLE.h` and one file in the nordic sdk is named `ble.h`.
If every folder is included that means that the `ble` folder is included as well  and `BLE.h` and `ble.h` will collide on case insensitive file systems.

As a **workaround**, `ble.h` in nordic SDK has been renamed `nrf_ble.h`.

I have updated all BLE library and featured examples repositories on `developer.mbed.org`. It should work for you now.

0xc0170, 2016-04-12:
I can confirm works for me ! Thanks @pan-

closed 2016-04-12 by 0xc0170