ARMmbed/mbed-os #1651: Add support for uVision 5 export

janjongboom, 2016-04-05:
It's no longer possible to open old uVision 4 projects in uVision 5 (see f.e. https://devzone.nordicsemi.com/question/31201/nrf51822-error-in-keil-device-not-found/), this now means that we're no longer compatible with latest uVision stable. We should fix that.

0xc0170, 2016-04-05:
> It's no longer possible to open old uVision 4 projects in uVision 5 

It is. There's legacy pack, but not valid for nrf51. You need to copy manually those files from nordic SDK (or they add those automatically, I recall I had to do it manually) to get device support in uvision5. 

Progen supports uvision5, even new uvision5 device packs for targets. There should be a migration path for this. We could add uvision5 support. People could use uvision4 or uvision5 (5 would default to new device packs, not the legacy device support). I can easily add this, however not many devices are in the progen definitions at the moment.

Or another option is to have uvision5 by default if target is supported there (progen has all data required), generates .uvprojx (=uvision5), otherwise tries to generate .uvproj.

I am inclined do go with new uvision5 exporter, start with small number of devices, and expand by the time.

janjongboom, 2016-04-05:
So opening a FRDM-K64F export to uVision 4 in uVision 5 ends with uVision crashing. (uVision IDE has stopped working, bla bla bla)...

0xc0170, 2016-04-05:
That never happened to me, it should not crash I assume.. I am using by default uvision5. Current version 5.17.0.0. 

I'll do a cleanup for the template project (I noticed there are some leftovers left), I added already exporter yaml file to get info about exporters, should be on master. Please try to generate blinky example for k64f, it should contain yaml file with more information about exporters. Share that with us, also uvision5 version.  I can have a look why a project crashes uvision5

toyowata, 2016-04-06:
@janjongboom @0xc0170 

You can also contact to Keil support team about IDE crash issue which is possibly a bug in the uvision.

I had an IDE crash case with latest version of uvision 5 IDE which worked fine with previous update version.  That was a clearly a bug in the IDE and Keil support team quickly sent me a fixed version for it.

BlackstoneEngineering, 2016-04-06:
This gets at a bigger issue, namely switching the exporters on mbed.org to use progen. Also, just FYI the nordic boards have a long history of being problem children with Keil.

0xc0170, 2016-04-06:
> This gets at a bigger issue, namely switching the exporters on mbed.org to use progen. 

More details? What issue is there that is bigger than this reported issue? uvision exporter which was not here was not uvision5 at all (there were some devices using uvision4 with uvision5 device packs). And progen is not yet enabled in the online IDE for uvision4 :wink:

btw, sarah is gone, she has not replied to any of the issues related to her changes for some time.

BlackstoneEngineering, 2016-04-06:
The issue isn't adding support for uvision v5 to the existing system, the issue is switching to progen, which will bring this implicitly. 

@0xc0170 , I am aware, but she will be back this summer.

janjongboom, 2016-04-07:
@toyowata Thanks. I opened an issue with Keil.

janjongboom, 2016-04-07:
FRDM-K64F issue has been resolved with latest patch from uVision.

0xc0170, 2016-04-07:
> FRDM-K64F issue has been resolved with latest patch from uVision.

Patch? U received something or just updated uvision? What was the cause?

We should either create a new feature request for uvision5 here in mbed, or rename this one. To get nrf51 for uvision5, we need to add uvision5 exporter to mbed exporters first.

janjongboom, 2016-04-07:
It was fixed in 5.18a, I was on 5.18.

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-