ARMmbed/mbed-os #1655: Problems with LPCXpresso export

janjongboom, 2016-04-06:
https://developer.mbed.org/questions/68593/Offline-toolchains-CodeRed-versus-LPCXpr/

0xc0170, 2016-04-13:
I am not much familiar with lpcxpresso project. We should have a quick look why it fails to compile. @janjongboom want to help with this one?

JojoS62, 2016-04-13:
I've tried also to export a LPC1768 project to LPCXpresso and it works fine. The code in the issue from R. Plant shows a linker error with undefined 'fprintf'. I find this only in older mbed versions in api/error.h, so I guess that the project just needs a mbed lib update. Or the project uses an (obsolete?) define for DEVICE_STDIO_MESSAGES.
old error.h:
`#if DEVICE_STDIO_MESSAGES
    #include <stdio.h>
    #define error(...) (fprintf(stderr, __VA_ARGS__), exit(1))
#else
    #define error(...) (exit(1))
#endif 
`

0xc0170, 2016-04-14:
@JojoS62 Thanks  for sharing. I'll ask for the mbed lib version

sg-, 2017-01-16:
Confirmed to be resolved

closed 2017-01-16 by sg-