ARMmbed/mbed-os #1657: FRDM-K64F: incorrect entry in PinMap_UART_RX

pythontech, 2016-04-11:
The code has PTC6 as an option for UART_0 RX, but it should be PTD6.  This can be verified from section 10.3.1 "K64 Signal Multiplexing and Pin Assignments" in the [K64 Sub-Family Reference Manual](http://cache.nxp.com/files/microcontrollers/doc/ref_manual/K64P144M120SF5RM.pdf)

```
diff --git a/libraries/mbed/targets/hal/TARGET_Freescale/TARGET_KPSDK_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.c b/libraries/mbed/targets/hal/TARGET_Freescale/TARGET_KPSDK_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.c
index f9a4db1..92009d7 100644
--- a/libraries/mbed/targets/hal/TARGET_Freescale/TARGET_KPSDK_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.c
+++ b/libraries/mbed/targets/hal/TARGET_Freescale/TARGET_KPSDK_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.c
@@ -108,7 +108,7 @@ const PinMap PinMap_UART_RX[] = {
     {PTC3 , UART_1, 3},
     {PTC14, UART_4, 3},
     {PTD2 , UART_2, 3},
-    {PTC6 , UART_0, 3},
+    {PTD6 , UART_0, 3},
     {NC  ,  NC    , 0}
 };
 
```

0xc0170, 2016-04-12:
Thanks for reporting. Please send a pull request with this change. I checked, PTC6 does not have UART mux option available, PTD6 does.

pythontech, 2016-04-16:
PR submitted: #1675

sg-, 2017-01-16:
@pythontech Looks like the fix has been merged

closed 2017-01-16 by sg-