ARMmbed/mbed-os #1663: Error in lwip sys_arch.c

Dieter-Graef, 2016-04-13:
Issue: sys_arch.c error

There is an error in the message fetching procedures that prevents STM devices from working.The current version does not cover the osOK event status. A corrected version for sys_arch_mbox_fetch is this:

u32_t sys_arch_mbox_fetch(sys_mbox_t *mbox, void msg, u32_t timeout) { u32_t start = us_ticker_read();

osEvent event = osMessageGet(mbox->id, (timeout != 0)?(timeout):(osWaitForever));

ifdef LWIP_DEBUG if (event.status!=1) { LWIP_DEBUGF(SYSARCH_DEBUG, ("mbox 0x%x %d\r\n",mbox,event.status)); }
endif if (event.status == osOK) {
msg = (void*)event.def.message_id; LWIP_DEBUGF(SYSARCH_DEBUG, ("sys_arch message 0x%x id %d \n",msg)); return (us_ticker_read() - start) / 1000; }
if (event.status == osEventMessage) { LWIP_DEBUGF(SYSARCH_DEBUG, ("sys_arch mbox fetch mbox 0x%x id %d \n",mbox,mbox->id));

msg = (void *)event.value.v; return (us_ticker_read() - start) / 1000; } else { return SYS_ARCH_TIMEOUT; }
}
sys_arch_mbox_tryfetch needs to be handled the same way

I dont have a URL to testcase program since the STM32F746 Discovery EMAC driver requires also changes in the linkerscript (only have gcc) to make use of SRAM2 ("The AHB masters support concurrent SRAM accesses (from the Ethernet or the USB OTG HS): for instance, the Ethernet MAC can read/write from/to SRAM2 while the CPU is reading/writing from/to SRAM1."-from the manual)

0xc0170, 2016-04-13:
Hi, your code snippet adds lot of debug info, it's hard to see what you have added. Can you highlight what you added?

0xc0170, 2016-04-13:
@geky @bogdanm

Dieter-Graef, 2016-04-13:
just this :
    if (event.status == osOK)
    {
    *msg = (void*)event.def.message_id;
    return ERR_OK;
    }

geky, 2016-08-01:
Hi @Dieter-Graef, if I'm reading the cmsis docs ([here](https://www.keil.com/pack/doc/cmsis/RTOS/html/group___c_m_s_i_s___r_t_o_s___message.html#ga6c6892b8f2296cca6becd57ca2d7e1ae)) correctly, osOK is the status code for when no message is available and the timeout is zero.

Looking at the lwip documentation ([here](https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip-sys/arch/lwip_sys_arch.c#L143)), lwip doesn't treat a timeout of zero as a special case and expects SYS_ARCH_TIMEOUT if no message is available.

It looks like the current implementation will correctly return SYS_ARCH_TIMEOUT when the status is osOK (osOK != osEventMessage). Is this not the case on STM devices?
``` c
osEvent event = osMessageGet(mbox->id, (timeout != 0)?(timeout):(osWaitForever));
if (event.status != osEventMessage)
    return SYS_ARCH_TIMEOUT;
```

LMESTM, 2017-05-29:
@geky @Dieter-Graef  Is this ticket still valid ? 

I tried to understand the possible link to ST targets, but this i not obvious.
The ST port layer will only call to netif_input_fn input callback when there is a message, then the intermediate layer will post the message to the mailbox. So if there is no data on the Ethernet link, ST porting layer will not post any message and the timeout would be fully handled in the OS layer. 

If I miss something, just let me know. If the problem is not applicable anymore, let's close.

LMESTM, 2017-06-29:
@0xc0170  @geky @Dieter-Graef Is this ticket still valid ?

geky, 2017-06-29:
I wasn't sure earlier, but since the rtx5 update we've completely rewritten the lwip's mbox implementation. The mbox was updated in this pr:
https://github.com/ARMmbed/mbed-os/pull/4203

So I believe this pr is no longer valid. I'll go ahead and close it. @Dieter-Graef feel free to reopen this issue if we're mistaken.

closed 2017-06-29 by geky