ARMmbed/mbed-os #1664: Gray border around <img/>'s in Safari

janjongboom, 2016-04-13:
All <img> tags in Safari have a little gray border around them...

[image]

There's some info in [this SO question](http://stackoverflow.com/questions/4743127/chrome-safari-display-border-around-image), but I think the quickest way would be to change to a div, because we set the image via CSS anyway it seems.

janjongboom, 2016-04-13:
Or add this to img CSS:

```
display: inline-block;
content: '';
```

0xc0170, 2016-04-13:
@stephenpaulger Where do we have a tracker for website? I believe it's on mbed.org in the forum, section bugs & suggestions, correct?

stephenpaulger, 2016-04-22:
That looks like an issue for @screamerbg to look at as it's in the IDE.

closed 2016-04-22 by stephenpaulger

janjongboom, 2016-04-22:
Tracked it down to uBlock. So if anyone else runs into this: disable uBlock :-)

It thinks /system/images/blank.gif?v=1234 is an ad, which is the blank image we're using as placeholders.